seemoo-lab/nexmon_csi #70: Error Installing Nexutil: unknown type name '_GL_INLINE_HEADER_BEGIN'

Junye97, 2020-05-10:
Hi there, 

I was trying to reinstall the nexmon csi on my R Pi 4B after an upgrade, however I got the following error when trying to compile nexutil: 

```
make install nexutil
cd ../libnexio && make
make[1]: Entering directory '/home/pi/nexmon/utilities/libnexio'
gcc -c libnexio.c -o libnexio.o -DBUILD_ON_RPI -DVERSION=\"IT_VERSION\" -I../../patches/include
ar rcs libnexio.a libnexio.o
make[1]: Leaving directory '/home/pi/nexmon/utilities/libnexio'
gcc -static -o nexutil nexutil.c bcmwifi_channels.c b64-encode.c b64-decode.c -DBUILD_ON_RPI -DVERSION=\"IT_VERSION\" -DUSE_NETLINK -I. -I../../patches/include -I../libnexio -L../libnexio/ -lnexio -I../libargp
In file included from nexutil.c:39:
../libargp/argp.h:583:3: error: #error "Please include config.h first."
  #error "Please include config.h first."
   ^~~~~
../libargp/argp.h:585:1: error: unknown type name '_GL_INLINE_HEADER_BEGIN'
 _GL_INLINE_HEADER_BEGIN
 ^~~~~~~~~~~~~~~~~~~~~~~
../libargp/argp.h:616:9: error: expected ';' before 'void'
 ARGP_EI void
         ^~~~
../libargp/argp.h:622:9: error: expected ';' before 'int'
 ARGP_EI int
         ^~~
../libargp/argp.h:634:9: error: expected ';' before 'int'
 ARGP_EI int
         ^~~
../libargp/argp.h:645:22: error: expected ';' before 'extern'
 _GL_INLINE_HEADER_END
                      ^
                      ;
make: *** [Makefile:15: nexutil] Error 1
```

Does that have anything to do with the following packages?

```
The following packages will be upgraded:
  binutils binutils-arm-linux-gnueabihf binutils-common bluez curl distro-info-data firmware-atheros firmware-brcm80211
  firmware-libertas firmware-misc-nonfree firmware-realtek fuse gir1.2-pango-1.0 git git-man gldriver-test iputils-ping
  libbinutils libbluetooth3 libcups2 libcupsimage2 libcurl3-gnutls libcurl4 libfm-data libfm-extra4 libfm-gtk-data libfm-gtk4
  libfm-modules libfm4 libfuse2 libgnutls30 libicu63 libjavascriptcoregtk-4.0-18 libldap-2.4-2 libldap-common libpango-1.0-0
  libpangocairo-1.0-0 libpangoft2-1.0-0 libpangoxft-1.0-0 libssh-gcrypt-4 libssl1.1 libtag1v5 libtag1v5-vanilla libvlc-bin
  libvlc5 libvlccore9 libwebkit2gtk-4.0-37 lxinput lxplug-cputemp lxplug-network lxplug-ptbatt lxterminal openssl pcmanfm
  pi-package pi-package-data pi-package-session pipanel pishutdown piwiz pprompt python-motephat python-pil python3-motephat
  python3-pgzero python3-pil raspi-config rc-gui realvnc-vnc-server rp-prefapps rpi-chromium-mods rpi-eeprom rpi-eeprom-images
  rpi-update tzdata vlc vlc-bin vlc-data vlc-l10n vlc-plugin-base vlc-plugin-notify vlc-plugin-qt vlc-plugin-samba
  vlc-plugin-skins2 vlc-plugin-video-output vlc-plugin-video-splitter vlc-plugin-visualization wpasupplicant
```

Thank you

jlinktu, 2020-05-10:
Can't reproduce.
Have you modified your sources? You should include argp-extern.h not argp.h in nexutil.c line 39.

Junye97, 2020-05-10:
Hi @jlinktu, 

Thank you for the help. I didn't change the source myself but I did try to install nexmon firmware on an Asus router from the same nexmon source on the R Pi.... Could this be the cause?

Anyway the issue is fixed and thank you.

jlinktu, 2020-05-10:
Yes, in step 10 of the getting started for the bcm4366c0 you change the file by running `sed -i 's/argp-extern/argp/' nexutil.c`.

closed 2020-05-10 by jlinktu