seemoo-lab/nexmon_csi #71: I can only receive a few UDP packets, and then it stops

wqiu-calgary, 2020-05-10:
Hi,

First I would like to thank all the contributors of this project.

I successfully followed the instructions  on a Raspberry Pi 4B. However when I issue the command tcpdump -i wlan0 dst port 5500 in a seperate window, I only receive about 20 UDP packets or so, then the packets stops. When issuing ifconfig wlan0 up again, I receives another 20 or so UDP packets again...

any idea what  I should I do next?  Thanks in advance.

jlinktu, 2020-05-11:
Hi,
your chip might be crashing.
Have a look at '/sys/kernel/debug/ieee80211/phyX/forensics' (substitute X with the current phy number). If it contains a crashdump after tcpdump stops receiving packets please post it here.
Also see https://github.com/seemoo-lab/nexmon_csi/issues/41#issuecomment-589268764.

wqiu-calgary, 2020-05-17:
Sorry for responding late. I tried the original enable monitor mode patch, and it seems working great. So I went back to the CSI patch again, followed all the steps, and had the same results: UDP packets stops after a while. Attached is the forensic file:
[forensics.txt](https://github.com/seemoo-lab/nexmon_csi/files/4639556/forensics.txt)

thanks again for the help.

closed 2020-05-21 by wqiu-calgary