seemoo-lab/nexmon_csi #73: RP4 crash when capturing CSI packets

wqiu-calgary, 2020-05-18:
Hi,
I have managed to collect CSI packets on a 80MHz bandwidth channel. However, after a while (a few minutes) the packets stopped. 

Note: I have successfully applied the patches issue#41 has suggested, and was able to capture 20MHz CSI packets for more than 12 hours on a 2.4Ghz channel.

I have included the forensics file:
[forensics2.txt](https://github.com/seemoo-lab/nexmon_csi/files/4645892/forensics2.txt)

your help is appreciated.

mzakharo, 2020-05-18:
It seems my patch may be incomplete. A 'return' maybe missing right after https://github.com/seemoo-lab/nexmon_csi/blob/82161c4199d112aedae5fb4ec2d3f191ff387e4c/src/csi_extractor.c#L203

wqiu-calgary, 2020-05-18:
thanks mzakharo, I will try this as soon as I get a chance.

wqiu-calgary, 2020-05-19:
Hi mzakharo,
do you also need a return in the following code? 
if (p_csi != 0) {
                printf("unexpected new csi, clearing old\n");
                pkt_buf_free_skb(osh, p_csi, 0);
            }

mzakharo, 2020-05-19:
no

wqiu-calgary, 2020-05-19:
right, we don't need it so i have already deleted the post. thanks

closed 2020-05-21 by wqiu-calgary