seemoo-lab/nexmon_csi #75: cannot find `tcpdump` tool on Asus RT-AC86U

yujianyuanhaha, 2020-05-26:
hi,

i am following the instructions and pass all step in **Getting Started**, and step 1-5 in **Usage**, while for the final step 

_Collect CSI by listening on UDP socket 5500, e.g. by using tcpdump: tcpdump -i wlan0 dst port 5500. There will be one UDP packet per configured core and spatial stream for each incoming frame matching the configured filter._

i cannot found `tcpdump` installed anywhere on Asus RT-AC86U router, there is one in local ubuntu machine, but i assume we should run  `tcpdump` on the router.

where can I find the command, or how can I install it on a router?

Best Regards.

YaohuaLee, 2020-05-29:
`Tcpdump `can be cross compiled into the router's needs, then uploaded to the router with `scp`.If you have trouble compiling, you can download tcpdump directly from the Merlin-tool

yujianyuanhaha, 2020-05-29:
> `Tcpdump `can be cross compiled into the router's needs, then uploaded to the router with `scp`.If you have trouble compiling, you can download tcpdump directly from the Merlin-tool

Got it.
 
Now I start over by install MERLIN rather than official firmware, in the way of:
1. install MERLIN
2. install entware
3. install opkg
4. install tcpdump

closed 2020-05-29 by yujianyuanhaha