seemoo-lab/nexmon_csi #78: 0 packets captured, tcpdump on NEXUS 5

yujianyuanhaha, 2020-05-29:
greetings,

I am following the instructions to extract CSI on nexus 5.
Everything works well, i pass all step in **Getting Started** bcm4339 , while in final step, i find `tcpdump` cannot capture any packages after running long time (and I ctrl+c to manually stop it).

```
root@hammerhead:/data/local/tmp # tcpdump -i wlan0 dst port 5500 -vv
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C
0 packets captured
6 packets received by filter
0 packets dropped by kernel
```

Additionally information:
1. the phone will lost connection to a specific AP in monitor mode, after run command 
```
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==
nexutil -Iwlan0 -m1
```

2. if I skip the **dst port**, like ` tcpdump -i wlan0 `,  I can successfully capture a lot of packets, however, these packets do NOT look valid, and `/nexmon_csi/utils/matlab.m` will tell error _skipped frame with incorrect size_, which means its WRONG to skip dst port.

3. **nexutil -k**
 `chanspec: 0xe29b, 157/80`

4. **ifconfig**
`wlan0     Link encap:Ethernet  HWaddr CC:FA:00:EB:A0:99
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2628173 errors:0 dropped:1995553 overruns:0 frame:0 
          TX packets:607 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:125554283 TX bytes:125696`

zeroby0, 2020-05-30:
Heyo!

Are you using `-vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==` in your actual script or the output generated by makecsiparams? The parameter you posted includes a Mac filter for mac IDs `00:11:22:33:44:55,aa:bb:aa:bb:aa:bb`, so if you are indeed using it, that's why you didn't get any output.

Run `makecsiparams -c 157/40 -C 1 -N 1` and use it's output instead. The `-b 0x88` is a filter that only allows packets starting with `0x88` (Router Beacons). You can omit the filter to capture all packets on  air. 

Also make sure there is a router that's broadcasting on [channel 157 (5GHz)](https://en.wikipedia.org/wiki/List_of_WLAN_channels). Use bandwidth 40, as shown in the command above. If 80 MHz channels on channel 157 are not allowed in your country, then the radio might not work. I just set my router and Pi to US in WiFi country.

Yes, you shouldn't skip dst port. Tcpdump, by default, captures all packets going through an interface. Packets sent by Nexmon are sent on UDP port 5500, and using the `dst port 5500` makes tcpdump save only packets going through port 5500.

Yes, the connection to AP is lost in monitor mode. WPA related software messes with the CSI collection, so they turn it off, and the connection is lost. If `wpa_supplicant` exists in Android (not sure it does, but google says it does), try turning it off.

Also check if the battery is able to power your device sufficiently; my raspberry failed to collect data when I was underpowering it.

Good luck! And post here if you got it to work. Also, please don't @ mention people unless they are already part of the discussion.

yujianyuanhaha, 2020-05-31:
@zeroby0  thanks a lot, your reply does solve my problem, now i can capture the CSI now !
I run `makecsiparams -c 157/40 -C 1 -N 1` and use it's output instead.

zeroby0, 2020-05-31:
You have to change `csireader.m` to your chip and bandwidth settings.

https://github.com/seemoo-lab/nexmon_csi/blob/aa370c6fb47c8a66c0958cbaacf0472f31189b16/utils/matlab/csireader.m#L12-L15

These should work for your Nexus 5.
```
 CHIP = '4339';              % wifi chip (possible values 4339, 4358, 43455c0, 4366c0) 
 BW = 40;                    % bandwidth 
 FILE = './PathToYour.pcap'; % capture file 
 NPKTS_MAX = 1000;           % max number of UDPs to process 
```

After you do that, you may still have a few incorrect size packets. I have not used a Nexus, but with my Pi, I do get incorrect frame sizes frequently. Almost half of my data is padded with `0`s, which increases the payload size from 572 bytes to 576 bytes (at 40 MHz). This increase is detected as incorrect size. I found the data to be usable nevertheless.

[image]

I recommend you download [Wireshark](https://www.wireshark.org/#download) and inspect the packets and the data they have. In wireshark, if you doubleclick one of the packets, it opens up the packet for inspection.

[image]

If you see Nexmon in the data, you have the correct packets, and can use them without worrying about the padding. Let me know if you still don't get at least once correct packet.

yujianyuanhaha, 2020-05-31:
@zeroby0  thanks a lot for your kind reply.
I change the bandwidth from 80 to 40, now I can see the CSI plot now !
I also install the wireshark and did see the nexmon in the data
thanks a again!

closed 2020-05-31 by yujianyuanhaha