seemoo-lab/nexmon_csi #79: __nex_driver_io: run nexutil error on ASUS RT-AC86U

yujianyuanhaha, 2020-06-02:
Greetings,

I meet an error when execute `./nexutil -Ieth6 -s500 -b -l34 -vSTRING`, where the STRING come from run `./makecsiparams -c 157/80 -C 1 -N 1 -m  -b 0x88`, then I see errror `__nex_driver_io: error`.
I got one time it runs without error, and successfully got the .pcap data, I may miss some step to repeat that

Some other information:
1. in ASUS-MERLIN configuration page, I manually pick **5GHz**, channel **157**, bandwidth **80M**. which matches the output of `./nexutil -Ieth6 -k`, and that of `./makecsiparams`.
2. I did turn the radio on by
```
wl -i eth6 up
wl -i eth6 radio on
ifconfig eth6 up
```
3. I also change some other channel-bandwidth setting, like 2.4G channel 11 bandwidth 20MHz, or 5G channel 153 40MHz, but they also 

Let me know if I need to provide other information. @jlinktu @zeroby0

Best Regards.

YaohuaLee, 2020-06-02:
Need to reload nexmon_csi firmware after reboot your router.

zeroby0, 2020-06-02:
I see a -m with no parameter to it. Try removing it.

Also, make sure that 157/80 is an allowed channel in your country. 

Do you get error with 157/40? 

Please post your exact commands.

yujianyuanhaha, 2020-06-02:
thanks a lot, @YaohuaLee the reload did fix the problem. I can get packets now!

thanks @zeroby0 as well

closed 2020-06-02 by yujianyuanhaha