seemoo-lab/nexmon_csi #80: Supporting more than 4 antennas

maxedbeech, 2020-06-03:
Hey!

Thanks a ton for such a fantastic repo. I have managed to get Nexmon working with a router with 8 antennas (Asus GT-AC5300), however I am currently not sure how I can get data back from more than 4 of those antennas.

Trying to run `makecsiparams` with C greater than 15 returns an `Invalid core mask` error. I have tried editing `makecsiparams` to prevent this check, however then the utility just returns the same value that it would if C was set to 15.

Does anyone have any ideas for how the utility can be edited to make this work please?

Thanks a lot!

mzakharo, 2020-06-03:
D11 microcode patch currently supports only cores 0-3: https://github.com/seemoo-lab/nexmon_csi/blob/aa370c6fb47c8a66c0958cbaacf0472f31189b16/src/csi.ucode.bcm4366c0.10_10_122_20.patch#L142

Code after this line needs some  work with cores 4-7.

For starters,  you would need definitions for ```ACPHY_TBL_ID_CORE{X}CHANESTTBL``` where {X} is 4-7. Looks like existing are 32 apart, so I would try  this:
```
#define		ACPHY_TBL_ID_CORE4CHANESTTBL	(169+32)
#define		ACPHY_TBL_ID_CORE5CHANESTTBL	(169 +32*2)
#define		ACPHY_TBL_ID_CORE6CHANESTTBL	(169+32*3)
#define		ACPHY_TBL_ID_CORE7CHANESTTBL	(169 + 32*4)
```

Have fun with d11 assembly. I found this page very helpful:
https://github.com/seemoo-lab/bcm-public/blob/master/buildtools/b43/assembler/test.asm

mikeWangJC, 2020-06-04:
@maxedbeech good job buddy !
Have you successfully run and collected data?

maxedbeech, 2020-06-04:
Great thanks a lot @mzakharo! Haha yup looks pretty daunting but will have a go!

@mikeWangJC Yes I have! @jlinktu provided some useful assistance (though of course only from 4 antennas so far)

maxedbeech, 2020-06-04:
@mzakharo it seems like there are two distinct problems here. Do you agree with this summary or am I missing something please?

### 1. Enable the makecsiparams utility to provide an output for when 16 <= C <= 255
- [X] Modify the IF statement that checks whether C <= 15
- [ ] Do some other (as yet unknown) work to get `makecsiparams` to provide a correct output for when C >= 15, instead of just providing the same output as if C = 15.

### 2. Modify Nexmon's d11 assembly code to support cores 4-7
- [X] Add definitions for `ACPHY_TBL_ID_CORE{X}CHANESTTBL` where {X} is 4-7
- [ ] Do some other (as yet unknown) work to the `csi.ucode.bcm4366c0.10_10_122_20.patch` file

jlinktu, 2020-06-04:
I think the 8 antennas are split up to two chips. According to https://deviwiki.com/wiki/ASUS_GT-AC5300 there are 3 4x4 capable SoCs (1 for 2.4Ghz and 2 for 5Ghz). If you can address them separately you should be able to capture CSI on 8 antennas by capturing on both 5 GHz SoCs at the same time. Thus, if both chips are exposed via an interface each just do the CSI extractor configuration and capturing for both interfaces.

mzakharo, 2020-06-04:
That sounds much easier! However, you do need to identify which interface what UDP packet came from, and I do not believe such identifier exists at the moment. One way this can be addressed is to add a 'tag' byte to the microcode payload, set by the user through 'makecsiparams'. Alternatively, maybe allow DST_PORT to be something other than 5500 for different interfaces.

maxedbeech, 2020-06-05:
Ah that's really smart @jlinktu, thanks a lot! I've played around and it looks like the available interfaces are:
- eth6: 2.4GHz
- eth7: 5GHz
- eth8: Unknown

eth8 is likely the interface we are looking for, however I haven't been able to get any packets retrieved via it yet (tried with both 5GHz and 2.4GHz transmitters). I'll keep playing around but if anyone has anything to suggest trying, would love to hear it as I'm fast running out of ideas.

Yes great ideas @mzakharo, thank you. Will look into those solutions once the packets start flowing...

maxedbeech, 2020-06-05:
Ah that's really smart @jlinktu, thanks a lot! So far what I think is the correct interface is not returning data, but I'll keep playing around with it and report here if I am able to make any progress.

Yes great ideas @mzakharo, thank you. Will look into those solutions once the packets start flowing...

maxedbeech, 2020-06-07:
So have done a fair number of tests and here is what I've found:
- For all the 5GHz channels tested between 32 and 64, only the first 5GHz interface worked
- For all the 5GHz channels tested between 100 and 128, only the second 5GHz interface worked

@jlinktu, @mzakharo (or anyone else), do you have any ideas on a workaround please?

mzakharo, 2020-06-08:
Not knowing for sure without taking this thing apart, this is likely a hardware limitation (as with many routers that use this approach). Even though chipset is the same on these interfaces, the RF front-end is likely different.  Each front-end is optimised for a different  5Ghz sub-band, yielding higher interference rejection, performance, etc.

jlinktu, 2020-06-08:
Have you checked if the interfaces both accept the chanspecs you are trying to set (e.g. with `nexutil -I<interface> -k` or `wl -i <interface> chanspec`)? This might be solvable by allowing additional chanspecs similar to as it's done for the bcm43455c0 and bcm4339 in `regulations.c`.

maxedbeech, 2020-06-09:
Many thanks for the thoughts @mzakharo and @jlinktu, very insightful.

@jlinktu I have tried modifying the `regulations.c` file, however sadly this has had no impact on the lack of packets being retrieved. Here are the results FYI (note they weren't executed in this order):

- `-c 36/80` & `nexutil -Ieth7 -k` => `chanspec: 0xe02a, 36/80` & packet retrieval success
- `-c 36/80` & `nexutil -Ieth8 -k` => `chanspec: 0xd064, 100` & packet retrieval fail
- `-c 100/80` & `nexutil -Ieth7 -k` => `chanspec: 0xe02a, 36/80` & packet retrieval fail
- `-c 100/80` & `nexutil -Ieth8 -k` => `chanspec: 0xe06a, 100/80` & packet retrieval success
- `-c 128/80` & `nexutil -Ieth7 -k` => `chanspec: 0xd024, 36` & packet retrieval fail
- `-c 128/80` & `nexutil -Ieth8 -k` => `chanspec: 0xe37a, 128/80` & packet retrieval success

So looks like this one is not going to be solvable, very much appreciate everyone's help anyway.