seemoo-lab/nexmon_csi #82: AoA Estimation based on the collected CSI

mikeWangJC, 2020-06-04:
Greetings!

Thanks somuch for the author to share such an excellent tool.
 
I have successfully extracted CSI from ASUS and tried to estimate AOA based on the extracted CSI.

However, during this process I meet a big problem.  I try to use a power splitter to measure the phase difference between tunnels, but the result indicates this phase difference is unstable, which is quite different from that of Intel CSItool, so the AOA estimation attempt failed.

Hope someone can help me or give some advice to fix this problem. 
Thanks in advance.

YaohuaLee, 2020-06-04:
I have the same problem. When using power divider to collect CSI, the phase difference of the same carrier between different packets seems very unstable. Some of the phase differences between successive packets are very stable, which is also conducive to phase cancellation. But there will also be some continuous phase disorder (up to a few hundred packets).

I filtered these packets, and used the phase difference between the packets that I thought was better for phase correction. After correction, their phase looks normal.

But after I collected CSI for AOA estimation, there was always angle difference. I wonder if the captured packet phase is very unstable, which leads to my error in phase cancellation. Finally, the AOA estimation also has problems.

mikeWangJC, 2020-06-15:
I've change the way to collect CSI.
However, this problem does not seem to be resolved.

tweigel-dev, 2020-12-01:
@mikeWangJC  which method for gathering CSI for AoA you are using now ?

tweigel-dev, 2020-12-02:
@mikeWangJC   Did you solve the problem or find a work around?
What did you mean with  power divider?  Did you use a piece of hardware like 
[image]
to measure the phase shift manually with an oscilloscope or do I understand you wrong?
by the way: may measurement are also very unstable and there is always a csi max at 32 at 40MHz
[image]

yujianyuanhaha, 2021-03-22:
Did you mean the [spotfi tool](https://web.stanford.edu/~skatti/pubs/sigcomm15-spotfi.pdf), which is open-sourced @mikeWangJC @tweigel-dev

zeroby0, 2021-03-22:
Isn't there an Automatic Gain Control in the WiFi chip which applies gain? That would change the estimated angle, right?

@yujianyuanhaha Spotfi needs agc gain too, but it's not available with nexmon_csi yet.
https://github.com/yuehanlyu/Wifi-Localization/blob/master/load_csi_data.py#L28

yujianyuanhaha, 2021-03-24:
@zeroby0  thanks, I also have atheros tool setup. But it disappoint me when I see the eventually accuracy is about median error 40cm.