seemoo-lab/nexmon_csi #84: split the CSI collect from 4 antennas on ASUS RT-AC86 Router

yujianyuanhaha, 2020-06-06:
greetings,

thanks for the advices along the way, I have been able to collect CSI from ASUS RT-AC86 Router now.

However, I am not clear how to split the CSI collect from 4 antennas, i used the encode string from `makecsiparams -c 157/80 -C 1 -N 0x0d  -b 0x88`, where the **0x0d** stands for 4 antennas. The [issue 51](https://github.com/seemoo-lab/nexmon_csi/issues/51#issuecomment-602220951) have mention the problem, has list to insert codes
```
    payloadbytes = typecast(payload,'uint8');
    core = bitand(payloadbytes(56),3);  % 
    rxss = bitand(bitshift(payloadbytes(56),-3),3);
```
after `payload = f.payload; ` of file nexmon_csi/utils/matlab/csireader.m.  However in my code both the `core` and `rxss` are all zeros after the operation.  Moreover, does the number 3 in the codes denote the 4th  attennas?  @matthiasseemoo @jlinktu

yujianyuanhaha, 2020-06-10:
i later figure out  the range of `core` is decide by the value you assign after `-C`, and `rxss` is decided by the one after `-N`. 
`core` could be understood as the number of antennas [reference](https://github.com/seemoo-lab/nexmon/issues/373#issuecomment-570942483), while `rxss` are the streams of that core.

closed 2020-06-10 by yujianyuanhaha

salmanpolito, 2020-07-06:
> i later figure out the range of `core` is decide by the value you assign after `-C`, and `rxss` is decided by the one after `-N`.
> `core` could be understood as the number of antennas [...]

Hey Jet Yu, Did you figure out how to separate the spatial streams? I have applied the recommended settings but it is not classified as four streams.