seemoo-lab/nexmon_csi #86: Can‘t make the nexutil about Nexus5

YaohuaLee, 2020-06-09:
I have successfully installed tools such as nexuti, but when I want to reload nexutil again, I fail. I don't know what happened, maybe I changed something in the middle, but I didn't find the reason for the failure. I hope someone can help me.

```
In file included from ./nexutil.c:39:0:
./../libargp/argp.h:583:3: error: #error "Please include config.h first."
  #error "Please include config.h first."
   ^
./../libargp/argp.h:585:1: error: unknown type name '_GL_INLINE_HEADER_BEGIN'
 _GL_INLINE_HEADER_BEGIN
 ^
./../libargp/argp.h:616:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 ARGP_EI void
         ^
./../libargp/argp.h:622:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'int'
 ARGP_EI int
         ^
./../libargp/argp.h:634:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'int'
 ARGP_EI int
         ^
./../libargp/argp.h:645:1: error: unknown type name '_GL_INLINE_HEADER_END'
 _GL_INLINE_HEADER_END
 ^
In file included from /home/yxy/android-ndk-r11c/platforms/android-21/arch-arm64/usr/include/byteswap.h:31:0,
                 from ./nexutil.c:41:
/home/yxy/android-ndk-r11c/platforms/android-21/arch-arm64/usr/include/sys/endian.h:52:10: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'htonl'
 uint32_t htonl(uint32_t) __pure2;
          ^
In file included from /home/yxy/nexmon/patches/include/asm/socket.h:19:0,
                 from ./../../patches/include/linux/socket.h:27,
                 from /home/yxy/android-ndk-r11c/platforms/android-21/arch-arm64/usr/include/sys/socket.h:34,
                 from ./nexutil.c:44:
/home/yxy/nexmon/patches/include/asm-generic/socket.h:21:65: fatal error: home/yxy/nexmon/patches/include/asm-generic/sockios.h: No such file or directory
 #include <home/yxy/nexmon/patches/include/asm-generic/sockios.h>
                                                                 ^
compilation terminated.
/home/yxy/android-ndk-r11c/build/core/build-binary.mk:460: recipe for target 'local/arm64-v8a/objs/nexutil/nexutil.o' failed
make[1]: *** [local/arm64-v8a/objs/nexutil/nexutil.o] Error 1
make[1]: Leaving directory '/home/yxy/nexmon/utilities/nexutil'
Makefile:11: recipe for target 'libs/armeabi/nexutil' failed
make: *** [libs/armeabi/nexutil] Error 2
```
Here are some of my environment

```
export NDK_ROOT=/home/yxy/android-ndk-r11c/
export PATH=$NDK_ROOT:$PATH
```

yujianyuanhaha, 2020-06-09:
I meet the problem before. you may try to reset the code by `git checkout .` under related directory.

YaohuaLee, 2020-06-11:
@yujianyuanhaha 
Thank you very much, sir. I'll try.

But now I have a question: what does spatial flow mean

For example, my transmitter has 4 antennas, and the receiver has 4 antennas. Does it mean that the maximum spatial flow is only 4, `- N 1 `means that I only receive data from the first antenna at the transmitter, `- N 3 `means that I receive data from the first and second antennas at the transmitter, `- N 7` means data from the first, second and third antennas?

yujianyuanhaha, 2020-06-12:
@YaohuaLee  my understanding is `-C` represent the index of chosen antennas, while `-N` is  the index of chosen  spatial stream per antenna.  @schmittner  may better answer the question

YaohuaLee, 2020-06-13:
@yujianyuanhaha 
Thank you for your answer. I think I get it.