seemoo-lab/nexmon_csi #87: can this project apply to mini-PCs？

hanzijun, 2020-06-10:
First thanks very  much for authors' contributions. I wonder wether this project can apply to mini-PCs？ And what is the max sampling rate？   thanks a lot :）

zeroby0, 2020-06-11:
If the WiFi chip used in your mini-pc is listed in the readme section, then it might work.

I believe the `-s500` in nexutil is the time delay between CSI calculations in microseconds. Setting a value less than this seems to cause instability. That would imply a theoretical limit of 2000 Hz for sampling. You should check the other issues and code to verify this.

matthiasseemoo, 2020-06-11:
-s500 is a SET (s) ioctl with number 500 and has nothing to do with the sampling rate:
https://github.com/seemoo-lab/nexmon_csi/blob/master/src/ioctl.c#L125 <https://github.com/seemoo-lab/nexmon_csi/blob/master/src/ioctl.c#L125>

The limiting factor for the sampling rate is sending up the dumped CSI information from the D11 core to the ARM firmware. CSI for 4x4 MIMO systems at 80 MHz bandwidth is much larger than a regular Wi-Fi frame, hence, the amount of frames you can process per second goes down.