seemoo-lab/nexmon_csi #88: cannot fully collect 40MHz or 80MHz bandwidth on ASUS RT-AC86U Router

yujianyuanhaha, 2020-06-11:
greetings,

Recently I notice that the ASUS router cannot fully collect 40M or 80M Hz bandwidth even I specific in the string as 157/40 or 157/80 in `makecsiparams` and later execute `nexutil` tool. For 40MHz it sometime use 40M while sometime use 20M, for 80M it sometime use 40M while sometimes use 20M but I never see 80M.

Some other information:
1. all command and matlab plot runs well without error, despite I **comment out** 
```
if f.header.orig_len-(HOFFSET-1)*4 ~= NFFT*4
        disp('skipped frame with incorrect size');
        continue;
    end
```
in file `./utils/matlab/csireader.m`

2. tcpdump can capture packages

zeroby0, 2020-06-11:
Heyo!

What you're seeing is normal. Here is my data, collected using 40MHz chanspec, but only 20MHz is filled.
[image]

Devices that can operate at 80 MHz routinely operate at lower bandwidths. This allows them to save power, make them compatible with other devices that can only operate at 40/20 MHz, etc.

You seem to be collecting csi with your router. You will need another device to inject frames. This device will let you create WiFi frames of whatever bandwidth you like. Make ping frames from this device and send to device you want to collect CSI from. The device will send back CSI response, which you can collect in your router.

[image]

There seems to be at least 2 ways to inject packets. One is with Nexmon, and the other is with airmon-ng. I don't know if it is easy with airmon, but with Nexmon, there are some hints in https://github.com/seemoo-lab/nexmon/issues/400 and https://github.com/seemoo-lab/nexmon_csi/issues/10.

I don't understand how to write an injector with Nexmon yet though. I'll probably use airmon-ng. In my case, I'm using a Raspberrypi to collect CSI from my router. I connected a laptop to WiFi, and sent ping to router, using normal ping command, no injection. That gave me 40MHz response.

matthiasseemoo, 2020-06-11:
> There seems to be at least 2 ways to inject packets. One is with Nexmon, and the other is with airmon-ng. I don't know if it is easy with airmon, but with Nexmon, there are some hints in seemoo-lab/nexmon#400 <https://github.com/seemoo-lab/nexmon/issues/400> and #10 <https://github.com/seemoo-lab/nexmon_csi/issues/10>.

Even if you are on a 40 or 80 MHz channel, you can send 20, 40 and 80 MHz frames. If you want to see 40 or 80 MHz frames, make sure that either your device really transmits them or that you inject them correctly.

zeroby0, 2020-06-11:
@matthiasseemoo 

This is a very frequent question in these issues. Would it be possible to write a guide on how to inject packets? Maybe write a python binding to inject packets from userland? Most people creating these issues are new to network programming, and it would be really helpful to have such a tool.

matthiasseemoo, 2020-06-11:
Of course, it would be possible, however, I do not have the time to do this. For the different chips, we have sendframe functions (see [1] for the bcm4339, for example, and [2] for the bcm43455) in the firmware and if the timer functions are defined, you can also inject frames periodically. And if you go through our code, you will find multiple example on how to create an ioctl that injects such frames, for example under [3]. There, you can also see how to correctly configure the rate for the different standards and the different bandwidths.

[1] https://github.com/seemoo-lab/wisec2017_nexmon_jammer/blob/master/src/sendframe.c <https://github.com/seemoo-lab/wisec2017_nexmon_jammer/blob/master/src/sendframe.c>
[2] https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43455c0/7_45_189/nexmon/src/sendframe.c <https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43455c0/7_45_189/nexmon/src/sendframe.c>
[3] https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489 <https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489>

zeroby0, 2020-06-11:
Thanks! If I get it working, I'll make a guide and make it available.

The CSI scanner cannot be used as a packet injector, is that correct?

matthiasseemoo, 2020-06-11:
Currently, it is not implemented, but it would make sense to add this functionality.

zeroby0, 2020-06-11:
Yes it's not implemented. Someone on the issues said the csi scanner should be only used for passively scanning, and a second device should be used for injecting ping frames. 

So I wanted to confirm if I can't inject frames while collecting CSI on same device.

matthiasseemoo, 2020-06-11:
You cannot inject the frames whose CSI you want to capture on the same device. However, you could take two devices and run the CSI extractor firmware and then use one for injection and the other for CSI extraction. However, the frame injection is currently not implemented in this firmware.

YaohuaLee, 2020-06-11:
@zeroby0 
I use the same method to collect CSI as you do, but I have two problems

(1) When setting the extraction parameters, it is necessary to set the number of antennas to extract CSI (for example `- N 11 = 1011`, which is equivalent to the 1, 2 and 4 antenna to collect CSI), but I can't distinguish the number of the 4 antennas, even if I use a power resolver. In MATLAB, their amplitude and phase can't see any difference at all (I connect three antennas of ac86u with power resolution , and then connect their antennas far away from ac86u). In this way, there is no AOA estimation or phase correction.

(2) When I use `tcpdump` to collect CSI, I can't collect any package after collecting a certain number of packages. I have to reload the firmware or restart ac86u (sometimes I can't even load the firmware).

How to solve it? I look forward to your help

zeroby0, 2020-06-11:
Hey @YaohuaLee 

I never used multiple antennas, so I can't be sure, but the data in Nexmon packets should have core and spatial-stream. 

> The next two bytes contain core and spatial stream number where the lowest three bits indicate the core and the next three bits the spatial stream number, e.g. 0x0019 (0b00011001) means core 0 and spatial stream 3. 

Maybe you can use that?

About 2, yeah that happens to me too. I restart and carry on. Usually this happens to me when the router is out of line of sight, or far away. This never happened to me in Line of Sight, but the number of packets I collect is less than 10,000. Someone said they face this when they collect about 40,000 packets.

@mzakharo found out there was a firmware trap a while ago (#41). Maybe it's related to that? There is a pull request from him which might fix the trap (#46). Please let me know if that fixes things for you.

3) I don't know what NSS is. Are you referring to the -N parameter of makecsiparams? It lets you set the spatial-stream. I don't know exactly how to use it, but @maxedbeech has used multiple antennas, and might be able to help you (#80).

I wish I could be of more help, but I'm a newbie myself XD.

yujianyuanhaha, 2020-06-11:
hi @YaohuaLee, few points regarding your question:
1. (a). there are 4 antennas in ASUS router, 3 is external and one is hidden, you can make full use of them by `-C 0x0f`.  (b). it is possible in 1x4 spatial stream or 4x4 spatial stream the amplitude does not change much if the transmitter and receiver is close to each other due to the channel quality is pretty good, you can try to move them apart to longer distance(or place sth between them) then each spatial stream should look different. 
2. (a) My transmit router is connect to internet, and a iPhone is playing video using this router, so there is a stable stream. As you can see this router is working as a normal router without been flashed.  Another router is used as receiver with nexmon firmware.  I do see when I stop playing youtube, the collecting of UDP slow down.  So I suggest at least you can try my way. Tho the injection the author mention is more professional.   (b). make sure `ssh HOST@192.168.0.x` always work before flash in firmware. I notice that I can switch off-and-on the wired connection under Ubuntu-Settings-Network, without shut-off-and-on router.

(3) My understanding is `-C` is the number of antennas, it range from 0x01 to 0x0f for ASUS router. `-N` is the spatial stream, it range from 0x01 to 0x0f for ASUS router, each antenna can receive at most 4 concurrent spatial stream at the same time.  e.g.  `-C 0x0f -N 0x0f` mean the receiver  use all 4 antennas, and each antenna  get 4 spatial stream at the same time.

YaohuaLee, 2020-06-12:
@zeroby0 @yujianyuanhaha 
Thank you very much.

As for the problem (1), the distance between my receiver and transmitter is relatively close, but I use the cable to combine one antenna of transmitter with one antenna of ac86u. The other three antennas of ac86u also use the cable to pull them far away from the transmitter (about 20 m, and there are two walls in the middle), but the collected signal strength is not very different Very strange.

As for question (2), I mean that after collecting a certain number of packets with `tcpdump`, there is no way to collect them again (I haven't counted the number, maybe tens of thousands, as zeroby0 said), and  must be reloaded the nexmon-csi firmware or restart the AC86u. As for the method mentioned by @mzakharo, it seems to be used on RI 4. I don't know if it works on ac86u. I will try it.

As for question (3), my transmitter has only two antennas, so when I set `-N  15`, where is the space flow coming from? At this time, as like as two peas of CSI, 4 sets of MATLAB appear in the CSI. The amplitude and phase of 4 CSI are identical, and 16 packets are 4 antennas simultaneously collected.

I will continue my research and communicate with you at any time.

matthiasseemoo, 2020-06-12:
Are you using attenuators between transmit and receive ports? If not, your signal levels might be very high and could even damage your receivers. 20 m of cable does not add a lot of attenuation, you will just observe a phase difference.

YaohuaLee, 2020-06-12:
@matthiasseemoo 
Thank you, sir. I will purchase the attenuator for testing. 
In addition, could you give me the advices to question (2), I am very troubled by the frequent collapse of the ac86u.

yujianyuanhaha, 2020-06-13:
i later figure out we need to check where channel is by check the router setting (type 192.168.0.x in browser), it could be channel 36, 100, 157 or others, set it from `Auto` to a constant, then tcpdump on that channel.
Also, `ping -i 0.1 ADDR` is good enough to get 80MHz bandwidth

closed 2020-06-13 by yujianyuanhaha

salmanpolito, 2020-07-06:
> @zeroby0
> I use the same method to collect CSI as you do, but I have 3 problems
> 
> (1) When setting the extraction parameters, it is necessary to set the number of antennas to extract CSI (for example `-C 11 = 1011`, which is equivalent to the 1, 2 and 4 antenna to collect CSI), but I can't distinguish the number of the 4 antennas, even if I use a power resolver. In MATLAB, their amplitude and phase can't see any difference at all (I connect three antennas of ac86u with power resolution , and then connect their antennas far away from ac86u). In this way, there is no AOA estimation or phase correction.
> 
> (2) When I use `tcpdump` to collect CSI, I can't collect any package after collecting a certain number of packages. I have to reload the firmware or restart ac86u (sometimes I can't even load the firmware).
> 
> (3) What does NSS mean? I set `- N (1 to 15)`, they can collect data (my transmitter only has two antennas), and it seems that the quality of CSI is not to much different. What's the matter?
> 
> How to solve it? I look forward to your help.

Hey YaohuaLee;
How did you manage to get multiple CSI streams from the given antenna? could you kindly shed some more light on it?