seemoo-lab/nexmon_csi #89: unknow mac address in UDP shown in Wireshark

yujianyuanhaha, 2020-06-11:
greeting,

Recently I can collect CSI information in .pcap file. Most of them looks right to me and the plot make sense. While the mac address of each UDP shown in Wireshark is bit confusing (see fig attached below)
[image]

The mac address `4e:45:58:4d:4f:4e` does not look belong to any of my mac address, which I thought should be target mac address I tcpdump on (or hear from).

Also, all command and matlab plot runs well without error, despite I comment out
```
if f.header.orig_len-(HOFFSET-1)*4 ~= NFFT*4
        disp('skipped frame with incorrect size');
        continue;
    end
```
in file `./utils/matlab/csireader.m`, not sure skip this warning matter or not.

zeroby0, 2020-06-11:
Nexmon collects all WiFi signals going in air, regardless of what router/device they came from.
Almost always, there's multiple routers operating in a single channel.
This unknown Mac ID you see is probably from your neighbour's router or some device nearby.

If you want to see CSI from only your router, you can set a filter on `makecsiparams` as shown in the readme. You can set allow upto 4 macIDs this way.

Even I have many unrelated Mac IDs in my data. I read pcap files in python, and remove mac ids I don't need.

matthiasseemoo, 2020-06-11:
It is just the mac address we set when we encapsulate CSI data in UDP/IP/Ethernet frames.

yujianyuanhaha, 2020-06-11:
@zeroby0 thanks, you are right, i find `4e:45:58:4d:4f:4e` spells 'NEXMON'  and my target mac address is exactly  after 11111111.

closed 2020-06-11 by yujianyuanhaha