seemoo-lab/nexmon_csi #90: How to slow down CSI collection?

zeroby0, 2020-06-12:
Currently I collect too many packets in a very short time, and the data is too much to process.

Is there a way to slow down CSI collection? Ideally I would like them to be averaged, failing that, a lower sampling rate too is good.

289536718, 2020-06-12:
@zeroby0 
Using the `ping` command can control the rate, and I can even `ping` a package in a second. You need to ensure that the transmitter is not accessed by other users, and that the computer used to send the request does not have a background program occupying the network speed.

zeroby0, 2020-06-12:
Hey Mignonv (@289536718),

That's a nice idea! But I don't have control over who accesses the transmitter. I'm trying to fingerprint CSI in an office setting, so I can't make changes.

I'm considering dropping csi samples from pcap files manually, or averaging them, but I wanted to see if there's a way to do that in Nexmon itself. Mostly because someone said the CSI calculations can also be statistical, so they'll be stabler and more 'correct' if given time.

closed 2020-06-13 by zeroby0