seemoo-lab/nexmon_csi #91: tcpdump on multiple channels on ASUS RT-AC86U Router

yujianyuanhaha, 2020-06-15:
greetings,

Is it possible to tcpdump on multiple channels on ASUS RT-AC86U Router at the same time?  I can run tcpdump on a single channel as in the demo and it works well.

My setup include 3 router work at different channel, e.g 36, 100, 157, and all at 80M bandwidth at 5GHz. Now I can tcpdump one-by-one on the 4th router , but not able to hear them simultaneously. Is there any advice or where to change the tcpdump source codes? @matthiasseemoo @jlinktu 

Best.

jlinktu, 2020-06-15:
That is not a limitation by `tcpdump` but the Wi-Fi chip itself. You can only tune it to one channel at the time, thus, you are not able to overhear several channels at once with only one chip. (Actually, that is not 100% accurate as e.g. 80MHz channels are just four bonded 20MHz channels, but that does not change anything in your case.)
What you can do is to iterate over the channels, this will result in missed frames but at least enables you to catch some on all three channels. For example, you can use `wl` or `nexutil` to switch channels while the extractor is running.
Another option is to use a device that comes with several Wi-Fi chips that can operate in the same band, e.g. Asus GT-AC5300 works with the patched firmware for the Asus RT-AC86U and features two 5GHz capable chips that can be used simultaneously, allowing to extract on two channels at the same time, however, the channeling seems to be currently limited (see #80).

yujianyuanhaha, 2020-06-15:
@jlinktu thanks for the reply. I may iterate over the channel way

closed 2020-06-15 by yujianyuanhaha