seemoo-lab/nexmon_csi #92: CSI definition at monitor mode

yujianyuanhaha, 2020-06-16:
In monitor mode, does the CSI present the link between the sniffer and the transmitter, or the link between the transmitter and corresponding receiver?

e.g. In the figure below, **router A** is communication with **device B**, while **sniffer router X**  is collect CSI in monitor mode. Does the CSI describe the **link AB**, or the **link AX**?
[image].

@matthiasseemoo @jlinktu @zeroby0  @289536718 I am aware it is a fundamental problem but I am not pretty sure...

zeroby0, 2020-06-16:
Link AX. CSI is measured using a known training sequence in WiFi frames. X compares the sequence it recieves to the how it should be, and determines CSI. CSI is just impulse response in a way.

yujianyuanhaha, 2020-06-16:
Got it, thanks bro~ @zeroby0

closed 2020-06-16 by yujianyuanhaha