seemoo-lab/nexmon_csi #93: CSI amplitudes for distance estimation (RBPi 4)

salomoneli, 2020-06-16:
Hi, I am currently implementing a simple distance estimation approach that uses the CSI amplitudes (i.e. the amplitude values for each subcarrier of one WiFi packet). I'm using the Nexmon CSI Extractor on a Raspberry Pi 4 to collect the CSI. As a transmitter I use another Raspberry Pi 4 that is sending WiFi frames using `ping`. The goal is to estimate the distance between these two Raspberry Pis.

The CSI collection works fine, but I'm facing two problems regarding the results: 
1.) Ideally, the _CSI amplitudes of several WiFi packets should stay constant over time_ at a certain distance. Of course this is not the case in real environments, but the CSI amplitudes of several WiFi packets, at the same distance, are too unstable.
2.) The _CSI amplitude values should decrease with an increasing distance_ between the two devices. But in my case the average of the CSI amplitude at a distance of 2m and 20m are almost equal.

I already tried several changes in the experimental setup, but nothing changed so far. 

So my questions are:
1.) Is anyone else facing a similar problem with unstable CSI values (amplitude and/or phase)?
2.) Has anyone already successfully implemented distance estimation using the Nexmon CSI Extractor and Raspberry Pis (or other Hardware) and maybe has a hint?

Thanks in advance!

mzakharo, 2020-06-16:
CSI waveform, obtained by nexmon,  goes through  [AGC](https://en.wikipedia.org/wiki/Automatic_gain_control) first, destroying most 'distance' information with it. To recover it, I have seen some researchers combine CSI measurement with RSSI. 
 Even at a stable physical distance, AGC will cause the CSI waveform to change, as the receiver tries to compensate for external blockers/interference/environment changes.

salomoneli, 2020-06-18:
Thank you so much for this hint!
Do you know if it is possible to get the gain that was applied for each packet?

wyhpolyu2020, 2020-06-18:
Dear Salomoneli, it seems that the csi tool works well on your raspberry pi B4. I am using this device too. I met a problem. When I execute the command pkill wpa_supplicant and ifconfig wlan0 up. I lost the wifi connection. So, my laptop cannot ping my raspberry 4 and when I collect the csi packet, it shows that 0 packet was collect. Would you mind give me some ideas? I really need your help.
Many thanks. Email- yuhao1995.wang@connect.polyu.hk

zeroby0, 2020-06-20:
@salomoneli 

https://arxiv.org/pdf/1910.06895.pdf

See section 4.1.3
[image]

@mzakharo has a patch #46 that has RSSI embedded in the same packet as CSI.

fatihoezdemir, 2020-07-01:
Hello @mzakharo , 
do you know how to get the RSSI for each packet for compensating these effects ?

salomoneli, 2020-07-01:
@mzakharo has written a patch that allows to get the RSSI together with CSI data.

Here's the link:
https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/7e3f9f720e1eb12ef11afd855515981d5a3b715b#

closed 2020-07-01 by salomoneli