seemoo-lab/nexmon_csi #96: Need help for the installation

p2146, 2020-06-18:
I plan to obtain CSI using Raspberry Pi 3B+/4. I am very new to both Raspberry Pi and also its firmware setting. Could someone please elaborate the steps for getting start?

I have read the getting started guide. I would like to know the steps before this i.e. 
before 1. Make sure the following commands are executed as root: sudo su 
Which program do I need to open to follow the guide and enter the commands? Which files do I need to download beforehand?

Thank you in advance.

zeroby0, 2020-06-19:
You need to enter it in terminal.

Wish I could make a longer guide, but I'm very busy till the 26th. Email me at aravind.reddy@iiitb.org and I can help you after that date.

p2146, 2020-06-19:
Thank you very much for your email.

I will email you after the 26th.