seemoo-lab/nexmon_csi #97: Utilizing this tool on Rasp pi 4B, no csi packet obtained

wyhpolyu2020, 2020-06-18:
Dear all, it seems that the csi tool could work on my rasp 4. I met a problem. When I execute the command pkill wpa_supplicant and ifconfig wlan0 up. I lost the wifi connection. So, my laptop cannot ping my raspberry 4 and when I collect the csi packet, it shows that 0 packet was collect. Would you mind give me some ideas? I really need your help.
In addition, when I use the 
makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88
command, there are two mac address should be changed, one I set my wlan0 address, what should be the other one? 

Many thanks. Email- yuhao1995.wang@connect.polyu.hk

zeroby0, 2020-06-19:
Hey,

You can't use the WiFi chip to connect to network when you're collecting CSI.

See #42. I use an Ethernet cable to ssh into the Pi.

`makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88`

You should remove both the Mac IDs, and put the Mac ID of the device whose CSI you want to collect. If you don't give any Mac ID, CSI of all the devices in that channel are collected. You should remove `-b 0x88` as well. If that flag is present, only CSI of packets that start with 0x88 is measured.

If you find that CSI collection gets stuck, see this pull request #46.

I may be slow to respond till this 22nd, but feel free to post here if you face any problems.

wyhpolyu2020, 2020-06-19:
Dear,

Thank you so much for answering me the question.
Now I can capture the csi packet but I cannot find where the system store the packet. I only can see the packet on the terminal window.
Do you have any idea on that?
Really appreciate for your help.

Best
Yuhao

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

Disclaimer:

This message (including any attachments) contains confidential information intended for a specific individual and purpose. If you are not the intended recipient, you should delete this message and notify the sender and The Hong Kong Polytechnic University (the University) immediately. Any disclosure, copying, or distribution of this message, or the taking of any action based on it, is strictly prohibited and may be unlawful.

The University specifically denies any responsibility for the accuracy or quality of information obtained through University E-mail Facilities. Any views and opinions expressed are only those of the author(s) and do not necessarily represent those of the University and the University accepts no liability whatsoever for any losses or damages incurred or caused to any party as a result of the use of such information.

zeroby0, 2020-06-19:
If you are following the readme, the tcpdump command given doesn't store the packets.

Use `tcpdump -i wlan0 dst port 5500 -c 100000 -w "location/filename.pcap" -vv`

`-c` is the number of packets you want to capture. `-w` is where your CSI is stored as a PCAP file. You can inspect pcap files with Wireshark.