seemoo-lab/nexmon_csi #100: CSI phase inversion and Jumping

YaohuaLee, 2020-07-07:
Hello，I encountered two problems when I used as86u to collect CSI .

1: As long as the incident angle of the signal is greater than a certain angle (estimated at about 35 degrees), the sign reversal problem will occur (positive sign becomes negative sign) during AOA.

[image]

2: The phase of the collected four antennas will jump suddenly, which seriously affects my positioning accuracy, but I can't solve it.
[image]

I tried a lot, but  couldn't solve the problem,
Hope someone can give me some advice. Thank you very much.

closed 2020-08-03 by YaohuaLee

yangtat, 2020-08-29:
Hello, I have the same problem as you. Have you solved it finally?

Deltaaaaaaaaaa, 2021-04-12:
@leeyewhwa Hello, Have you solved the phase jumping problem? Could you please give some clue?

yujianyuanhaha, 2021-07-29:
@leeyewhwa @yangtat I also meet the problem, I will treat these less appeared as outliers.

Meanwhile, did you  have the plot that  phase different between different antennas, are they nearly constant over subcarriers?

Best

Jet

yangtat, 2021-08-02:
@yujianyuanhaha The phase different between different antennas are nearly constant.
[image]

yujianyuanhaha, 2021-08-02:
@yangtat thanks, looks pretty good with down to 5-degree variations. Does this plot come after RF-chain calibration? Is the phase different offset constant the same for all subcarriers？

yangtat, 2021-08-03:
@yujianyuanhaha I just used the raw data collected by Nexmon. `phase_diff=angle(csi_11./csi_12)` csi_11 and csi_12 are the CSI matrix of the two receiving antennas.
I think the phase difference offset constant of different subcarriers is the same, but it seems to be different when the time between two acquisition is longer.

yujianyuanhaha, 2021-08-03:
@yangtat thanks
my results are like
[image]

yangtat, 2021-08-03:
@yujianyuanhaha I agree with your conclusion.
My previous plot is phase difference of one subcarrier over about 800 packets.
[image]
There seems to be  offset between the subcarriers in my image.
The device I use is a Nexus 6P

yujianyuanhaha, 2021-08-03:
@yangtat  thanks for sharing.

interesting, I also found the pattern in some antennas pair changes over time (altho previous plot 1th-5th does not)
[image]
This plot is phase difference over time for 3rd and 11th spatial steams, with 15 consective packages, the first 6 packages (in red) follow one pattern, while remain switch to another pattern.

yujianyuanhaha, 2021-08-05:
@leeyewhwa well, it looks to me the sudden jump is about 0.5 (from you plot is -0.15 to 0.35), it match the (**On 2.4 GHz bands, it appears that each antenna experiences an unknown phase shift by a multiple of π/2 (or 90 degrees)**.)[https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/6#issuecomment-77093589].

So you can switch to 5G to try again.