seemoo-lab/nexmon_csi #101: Which channel should I identify when I want to collect data on 40MHz, BCM43455c0, Rsp4B? -c ?/40

YuhaoWangPolyU, 2020-07-08:
I tried -c 1;2;3;4/40 but they did not work and said that invalid chan spec

salmanpolito, 2020-07-09:
> I tried -c 1;2;3;4/40 but they did not work and said that invalid chan spec

use ./nexutil -k<channel>/<bandwidth> to switch to another channel. for example, ./nexutil -k2/20

zeroby0, 2020-07-09:
https://en.wikipedia.org/wiki/List_of_WLAN_channels

Look at the 5 GHz or 5.8 GHz table.
[image]

If you want to collect csi from 40Mhz channel 46, use -c 44/40 or -c 48/40. Both are equivalent. You should collect from the channel in which your device of interest transmits.

haochenku, 2021-07-07:
THank you @zeroby0 , how about 157/80, does this mean it will collect csi on channel 155? 
[image]

zeroby0, 2021-07-07:
It does. You would also need to add the channel/bandwidth to the regulations.c file if you're using 157.

But even after that, I've only ever been able to reliably collect on channel 32. It's some bug in the broadcomm drivers, it affects other chips too, unless they are in channel 32.

haochenku, 2021-07-07:
@zeroby0   Thank you so much for your reply. Then I got confused on the data I collected. I donot use frame filter, ie., there is no -b parameter in my setup. I use 157/80 as channel spec. I cannot explain why I sometimes got very low csi values. Are they purely noise? why my data csi so less compared to noise? 
[image]

zeroby0, 2021-07-07:
You're welcome :)

They are probably from other routers or devices. Are you using the Mac filter to get only the packets from your router?

Also, even if you're using 157/80, and your router is set to transmit 80 mhz on 157, they transmit lower bandwidth packets all the time, so half of the graph looks like valid csi and the other half looks like the noise. But I don't think much of that is happening in your case.