seemoo-lab/nexmon_csi #102: nexus 5 wifi not work after flash in nexmoon tool

yujianyuanhaha, 2020-07-15:
greetings,

I have been able to successfully flash in nexmon into a nexu5 phone, and successfully collect the CSI data. However, I able not able to turn on the wifi anymore when I am not doing the CSI collection.

1. I flash into two phones, both of them can collect CSI, but then fail to connect to WiFi. I am aware the nexmon require the phone to run in monitor mode. I feel like the program does not jump out of monitor mode
2. I restart the phone, while still cannot connect to WiFi
3. I run command `ifconfig wlan0 up`, FAIL
4. I run command `svc enable wifi`, FAIL

Any ideas , thanks in advance.

mzakharo, 2020-07-15:
you need to reflash the original/stock firmware to get client wifi functionality back.

yujianyuanhaha, 2020-07-21:
@mzakharo 
thanks, anyway I can get wifi functionality back while still use nexmon?

gyy0926, 2020-09-15:
You can use Bluetooth to connect to the Internet.
Use another mobile phone or computer as a WiFi hotspot, and turn on the Bluetooth sharing network function.
Use nexus 5 to connect to the hotspot device via Bluetooth, and you can realize nexus 5's Internet access function. For detailed steps, you can search for the "Bluetooth Sharing Network" tutorial.

zeroby0, 2020-09-15:
@gyy0926 Have you tried it? I don't know if it carries over to Nexus, but Bluetooth and CSI collection interrupted each other in Raspberry Pi. If you find the same thing happening, an alternative is reverse USB tethering.

gyy0926, 2020-09-19:
Yes, I have tried it.
Turn on the "Share Network via Bluetooth" in a mobile phone A that can access the Internet. The AP frequency band is 2.4GHz.
Turn on the Bluetooth function of nexus 5 and connect to the Bluetooth of A mobile phone, check the "Internet connection" option.
I have tried it and can collect CSI data packets while browsing the web.
I didn't use Raspberry Pi, and I don't know the interruption between Bluetooth and CSI collection.

 

顾阳阳
gu_gao_yang@qq.com