seemoo-lab/nexmon_csi #103: RT-AC86U can't use tcpdump

13086675992, 2020-07-17:
greetings,

I am recently trying to collect CSI from RT-AC86U. When I execute the Step 6 command **tcpdump -i wlan0 dst port 5500**, the following message is displayed - SH: tcpdump: not found.I tried many of the methods mentioned in the question, but it still didn't work.I see that some friends have solved this problem with cross compilation, but they have not provided any available instructions. I am not familiar with the operation under Linux system, and I hope to get some direct and accurate guidance.
@matthiasseemoo @jlinktu 

best.

zeroby0, 2020-07-17:
See #75

13086675992, 2020-07-18:
> 如果还没有，请参阅[＃75](https://github.com/seemoo-lab/nexmon_csi/issues/75)
Thank you for your reply. I have the following problems when I follow the 75 question steps. What should I do next?
configure: error: C compiler cannot create executables
See `config.log' for more details.

zeroby0, 2020-07-25:
Where in #75 did you get this error? Try the last post in 75.

13086675992, 2020-08-28:
Thanks for your reply.I from https://github.com/RMerl/am-toolchains got the tcpdump file and copy it to the router. Then it can work normally.

jokaorgua, 2022-08-29:
@Dr-Yongqiang could you please share compiled version of tcpdump? Thanks in advance

zeroby0, 2022-08-29:
I have one here in [nexmonster-rtac86u.zip](https://github.com/nexmonster/nexmon_csi/files/5672773/nexmonster-rtac86u.zip).

https://github.com/nexmonster/nexmon_csi/discussions/2