seemoo-lab/nexmon_csi #104: Problem about generating traffic

huhanwj, 2020-07-23:
I successfully installed the tool on Raspberry Pi 4B but got a bit confused about the traffic generation. 
Should I try to generate traffic between my PC and the RPi or just traffic between my PC and my laptop/router on the specified channel?

zeroby0, 2020-07-23:
Send pings to your router IP via any wireless device connected to it the router. That should generate wireless traffic, and the CSI for packets in that traffic will be calculated by the RPi.

Also check out the pull request #46 if you're using Rpi and are experiencing problems with csi collection freezing.

huhanwj, 2020-07-30:
> 
> 
> Send pings to your router IP via any wireless device connected to the router. That should generate wireless traffic. The Rpi will calculate CSI for packets in that traffic.
> 
> Also check out the pull request #46 if you're using Rpi and are experiencing problems with csi collection freezing.

Thanks for your help!

closed 2020-07-30 by huhanwj