seemoo-lab/nexmon_csi #105: increasing the number of mac addresses to filter for

hyunhi2, 2020-07-24:
Hi,
thank you for the great work.
I just installed the tool on nexus 6P and it works well so far.
What I'm trying to do now is to increase the number of mac addresses to filter for.
Is it possible to increase the number or to mask part of mac address as in tcpdump?
If so, please give instructions on how to increase the number. I found ioctl.c parse command and write shared memory, but I am new to other codes.

Thanks in advance.

zeroby0, 2020-07-24:
I'd just filter using tcpdump or in post-processing instead of adding mac filters to nexmon.

Changing nexmon gets complicated quickly. You'd have to add new data structures to hold mac ids, and modify how chanspec generated by makecsiparams.c is handled.

hyunhi2, 2020-07-25:
You mean extracting csi of all packets first and filtering them with tcpdump?
I forgot the source mac address in payloads. It may work I guess.
Thank you!

closed 2020-07-25 by hyunhi2

zeroby0, 2020-07-25:
Yes, from the payload. The utils/matlab/ folder has examples. 

It works, I read the samples from the pcap file into a pandas dataframe and drop rows that are not in my list of Mac IDs. Do note that the Mac ID is in bytes. You would have to convert the hex string before you can compare.

hyunhi2, 2020-07-25:
understood. Thanks for your kind comments.