seemoo-lab/nexmon_csi #107: Can't make for Raspberry 4B+!!

suring-t, 2020-08-03:
In recent days, the kernel's version of is  **5.4.51** after "apt-get update && apt-get upgrade". So 'make' can't be continued in the following steps. Did anyone meet this problem?

joextodd, 2020-08-07:
What is the exact error message you are getting?

suring-t, 2020-08-08:
pi@raspberrypi:~ $ apt-get update && apt-get upgrade
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.51+ /boot/kernel.img
.......
root@raspberrypi:/home/pi# sudo reboot
root@raspberrypi:/home/pi# uname -a

Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux

root@raspberrypi:/home/pi/nexmon# cd patches/bcm43455c0/7_45_189/nexmon/
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon# make
fatal: not a git repository (or any of the parent directories): .git
............
APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.19.118-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:52: brcmfmac.ko] Error 2

joextodd, 2020-08-08:
I managed to get past this step with [rpi-source](https://github.com/notro/rpi-source/wiki). This will install the kernel headers, once installed the `/lib/modules/4.19.118-v7l+/build` directory will get symlinked to the appropriate headers directory.

However I think you need need to skip the `apt-get upgrade` step and stick with the default Raspbian kernel version 4.19, as the build will fail later down the line if it is not one of the [supported versions](https://github.com/seemoo-lab/nexmon#supported-devices).

huhanwj, 2020-08-13:
> What is the exact error message you are getting?
> 
> I skipped the `apt-get upgrade` step, and stuck with the supported kernel version of 4.14/19
> 
> You may need to install the kernel headers for this version using [rpi-source](https://github.com/notro/rpi-source/wiki)

You can also directly go to the Raspbian archive to get the older version of the kernel headers, simply going to https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware to find the one you need and use `dpkg` to install it.

jlinktu, 2020-08-28:
Added support for kernel 5.4 https://github.com/seemoo-lab/nexmon_csi/commit/34e6075c335e295248e0209d6dca38096230c3c1

closed 2020-08-28 by jlinktu