seemoo-lab/nexmon_csi #109: I encountered Some questions when compiling 'nexmon_csi'  folder

13086675992, 2020-08-15:
greetings,
I'm installing your firmware on the RTAC86U router.When I perform step 7, the following error message appears.
PATCHING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.patch => src/csi.ucode.bcm4366c0.10_10_122_20.asm
  ASSEMBLING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.asm => gen/ucode.bin
Makefile:215: recipe for target 'gen/ucode.bin' failed
make: *** [gen/ucode.bin] Error 1

I have tried many ways to solve these problems. I hope to get the help of my predecessors.
@qtux @schmittner @derfalx @DanielAW @Kjili 
Any ideas , thanks in advance.

matthiasseemoo, 2020-08-15:
did you run `make` in nexmon’s buildtool directory? Most likely you did not properly install the b43 assembler/disassembler.

You can also take a look into the log files to get more information about the problem.

13086675992, 2020-08-15:
> did you run `make` in nexmon’s buildtool directory? Most likely you did not properly install the b43 assembler/disassembler. You can also take a look into the log files to get more information about the problem.
Thank you for your reply. I look the log file under 'nexmon_csi' directory. 
1.error.log
cat: gen/nexmon.pre: No such file or directory
2.ass.log
gcc: error trying to exec 'cc1plus': execvp: No such file or directory
Assembler ERROR:
  No %arch defined
3.patch.log
patching file src/csi.ucode.bcm4366c0.10_10_122_20.asm
Hunk #7 FAILED at 5360.
Hunk #8 FAILED at 6835.
2 out of 8 hunks FAILED -- saving rejects to file src/csi.ucode.bcm4366c0.10_10_122_20.asm.rej

I operated through VMware and I don't know if this is the reason.
I don't know how to solve these problems. Can you give me some suggestions? thank you.

matthiasseemoo, 2020-08-16:
Did you manage to build the regular nexmon patch from nexmon.org <http://nexmon.org/>? it looks like you didnot execute setup_env.sh

Matthias

13086675992, 2020-08-18:
Thank you for your reply. I found the problem because I didn't install G + +. In addition, I hope that nexmon members will add the operation steps of installing make, pthon, G + +, how to obtain tcptump file and copy it to the router's file system, and copying the collected data file from rtac86u to the local host. In this way, beginners can quickly use this excellent project after getting the new machine installed with the Ubuntu system, which can save a lot of time. Thanks again！

closed 2020-08-19 by 13086675992

KyriacosStr, 2020-10-08:
Hi @13086675992 @matthiasseemoo

I have been running upon the same error while trying to install the firmware on Asus RTac86u.
I believe that I am executing correctly the instructions up until step no.7 where the same error message appears. (log files also portray the same lines as above..)

Any clues on how to proceed please??

Thanks in advance :D

[image]

KyriacosStr, 2020-10-08:
Hi @13086675992 @matthiasseemoo 

I have been trying to install the firmware on Asus rtac86u yet i run upon the same error message on step 7 and the log files contain the same lines as above..

[image]

I am following a similar approach as above since i am operating through VMware and I believe that I am correctly following the previous instructions since no errors appear during the 'make' process.

Any assistance would be greatly appreciated!

chivas1000, 2021-03-24:
@13086675992 did you successfully figure out the problem? I also encountered the same error, any help or advice would be much helpful, thanks.

DennisHerell, 2021-03-26:
@chivas1000 I also encountered similar error, but I'm using Nexus 6P. In my case, I got that error after failing to disassemble the ucode when trying to run it the first time, and the trying to run `make install-firmware` again. My first error is because I don't have python2 installed:
[image]

So I just installed python2, but the following error still come up:

[image]

 I solve the issue by starting over, deleting and download the nexmon_csi again. This time with no error in disassembling, there's no problem in installation. If you also get an error when disassembling, you can try what I did. If that's not the case, I'm also not sure since I'm not using the router.

Btw, you had some problem with the SDR right? Is there still some problem? sorry, I forgot to about it then

13086675992, 2021-03-26:
> @13086675992 did you successfully figure out the problem? I also encountered the same error, any help or advice would be much helpful, thanks.
Hi,I guess you don't have gcc, g++ and python installed. You can use the following commands to install these packages on the PC terminal.
sudo apt-get install g++
sudo apt-get install gcc
sudo apt-get install python

chivas1000, 2021-03-26:
>

@13086675992 HI There, that really is the gcc problem, i didn't got gcc installed, thanks for your help sincerely!

chivas1000, 2021-03-26:
> 
> 
> @chivas1000 I also encountered similar error, but I'm using Nexus 6P. In my case, I got that error after failing to disassemble the ucode when trying to run it the first time, and the trying to run `make install-firmware` again. My first error is because I don't have python2 installed:
> [image]
> 
> So I just installed python2, but the following error still come up:
> 
> [image]
> 
> I solve the issue by starting over, deleting and download the nexmon_csi again. This time with no error in disassembling, there's no problem in installation. If you also get an error when disassembling, you can try what I did. If that's not the case, I'm also not sure since I'm not using the router.
> 
> Btw, you had some problem with the SDR right? Is there still some problem? sorry, I forgot to reply about it then

@DennisHerell Hi, at this case my gcc not installed, so i might had to install as much dependencies next time lol, for the SDR, I think the problem is that I am using a RPI4B, since the make install command is
 **make install-rpi3plus**
I think it might uniquely support RPI3B+ only, I'm getting a 3B+ and prepare to try again, thanks for your detailed support.