seemoo-lab/nexmon_csi #111: Real-time csi plot on raspberry pi4

YuhaoWangPolyU, 2020-08-31:
Anybody have any idea that plot the tcpdump data into Matlab in real-time?

zeroby0, 2020-09-11:
With tcpdump? There are ways, but it's kinda messy.
I would use Python/Matlab to directly listen to the port in which CSI UDP packets are sent, and decode and plot them. You can also port forward that port to your laptop and do the plotting in your laptop.

YuhaoWangPolyU, 2020-09-20:
> With tcpdump? There are ways, but it's kinda messy.
> I would use Python/Matlab to directly listen to the port in which CSI UDP packets are sent, and decode and plot them. You can also port forward that port to your laptop and do the plotting in your laptop.

Dear, many thanks for your reply. As what you say, for example, if we use the example given in the readme, I just need to use Matlab to listen the port 5500 remotely and then I can get the csi packets?

1024610711, 2020-10-29:
> > 用tcpdump？有办法，但是有点混乱。
> > 我将使用Python / Matlab直接侦听发送CSI UDP数据包的端口，并对它们进行解码和绘制。您还可以将该端口移植到笔记本电脑，然后在笔记本电脑中进行绘图。
> 
> 亲爱的，非常感谢您的回复。正如您所说，例如，如果我们使用自述文件中的示例，我只需要使用Matlab远程监听5500端口，然后我就能得到csi数据包？

> 有人知道将tcpdump数据实时绘制到Matlab中吗？

Have you solved the problem?

YuhaoWangPolyU, 2020-10-29:
No...

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

Disclaimer:

This message (including any attachments) contains confidential information intended for a specific individual and purpose. If you are not the intended recipient, you should delete this message and notify the sender and The Hong Kong Polytechnic University (the University) immediately. Any disclosure, copying, or distribution of this message, or the taking of any action based on it, is strictly prohibited and may be unlawful.

The University specifically denies any responsibility for the accuracy or quality of information obtained through University E-mail Facilities. Any views and opinions expressed are only those of the author(s) and do not necessarily represent those of the University and the University accepts no liability whatsoever for any losses or damages incurred or caused to any party as a result of the use of such information.

1024610711, 2020-10-29:
> 我将使用Python / Matlab直接侦听发送CSI UDP数据包的端口

I also would use Python to directly listen to the port that sends CSI UDP packets, but why can’t I listen to any data on port 5500 with python?Can you give me some suggestions?

YuhaoWangPolyU, 2020-10-29:
Chinese?

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

1024610711, 2020-10-29:
Yeah！

 Chinese? 
 
 
 
 

 
 
 
 [https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

YuhaoWangPolyU, 2020-10-29:
你是想在rasp4 上面运行的python 还是远程到你rasp的啊

Yeah！

Chinese?

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

Arsering, 2021-09-14:
> Dear, many thanks for your reply. As what you say, for example, if we use the example given in the readme, I just need to use Matlab to listen the port 5500 remotely and then I can get the csi [...]

Hi, I have the same requirement as yours, which is to process the CSI in real-time. Do you know how to realize it?

Arsering, 2021-09-14:
> 你是想在rasp4 上面运行的python 还是远程到你rasp的啊 On Oct 29, 2020, at 4:25 PM, 1024610711 <notifications@github.com<mailto:notifications@github.com>> wrote: Yeah！
> […](#)

 @YuhaoWangPolyU Hi, I have the same requirement as yours, which is to process the CSI in real-time. Do you know how to realize it?

Zilliax-Barry, 2022-02-22:
Hi, is this problem solved? Cuz I have this same issue.

sarielkk, 2023-08-29:
> 您好，请问这个问题解决了吗？因为我也有同样的问题。

Hello， did you solve the problem? I have the same problem.

sie2, 2023-12-19:
This is probably a bit late, but you can try our solution [nexmon_csi_gui](https://github.com/ConnectedSystemsLab/nexmon_csi_gui/tree/main).

closed 2023-12-19 by jlinktu