seemoo-lab/nexmon_csi #112: split the 4 spatial stream CSI collected on NEXUS5

yujianyuanhaha, 2020-09-04:
have any one successfully split the 4 spatial stream CSI collected on NEXUS5? I am not able to. The code (work for ASUS) I used alway give me one spatial stream (all in zeors spatial stream index).

1. the code I use
```
 payloadbytes = typecast(payload,'uint8');
 core = bitand(payloadbytes(56),3); %
 rxss = bitand(bitshift(payloadbytes(56),-3),3);
 Nss(k) = core*4 + rxss;
```
, it works good for ASUS CSI

Cyrbaby, 2020-09-04:
NEXUS5 is 1x1 mimo，just have one spatial stream

matthiasseemoo, 2020-09-04:
That is not exactly true. You have only one antenna, still a transmitter can transmit with up to four antennas and the Nexus 5 can estimate the CSI to all four transmit antennas. You just need to inject frames that use multiple antennas for transmissions to stations that can only receive with one antenna. Or you patch the ucode in the Nexus 5 keep frames that are not receivable with only one antenna.

Cyrbaby, 2020-09-04:
> That is not exactly true. You have only one antenna, still a transmitter can transmit with up to four antennas and the Nexus 5 can estimate the CSI to all four transmit antennas. You just need to [...]

Thanks for the answer！I have misunderstood some aspects of spatial stream.May I ask another question?If using Asus RT-AC86U to acquire CSI, how to match the index of the core with the index of the physical antenna on the router Asus RT-AC86U?I have tried to  compare phase difference,but there are  many ways to arrange and combine the phase difference of different cores.Or how to exclude the core of the built-in antenna? Looking forward to more communication with you.

matthiasseemoo, 2020-09-04:
> … <x-msg://13/#>

Just remove all antennas and then install one by one and try to capture CSI. You should then see in which CSI plot you have some signal.

Cyrbaby, 2020-09-05:
Thank you for your answer. Your answer is very helpful to me

> Just remove all antennas and then install one by one and try to capture CSI. You should then see in which CSI plot you have some signal.
> […](#)

Thank you for your answer！It  is very helpful to me

yujianyuanhaha, 2020-09-07:
thanks, how can I patch the ucode? @matthiasseemoo 

my NEXUS and ASUS both receive the package from another same router at same time, the inject frame should be in expected way, which transmitting from multiple antenna.

yujianyuanhaha, 2020-10-09:
hi,

I look carefully into the [data format description](https://github.com/seemoo-lab/nexmon_csi#analyzing-the-csi) and location the byte.

Thanks a lot

closed 2020-10-09 by yujianyuanhaha