seemoo-lab/nexmon_csi #113: Unit of CSI amplitude

zmsenqn, 2020-09-07:
Hello,

I really appreciate for publishing Nexmon project on Github.

I was able to get CSI using a Raspberry Pi 3, but I have a question about the acquired CSI amplitudes.

The MATLAB code on this repository shows me CSI amplitude over sub-carriers, however, I am struggling to identify a proper unit (measurement) for the CSI amplitude. 

What kind of unit will be suitable for the CSI amplitude?

matthiasseemoo, 2020-09-09:
Hi,

CSI does not have a unit. Considering X a column vector of QAM symbols on one subcarrier transmitted using a number of spacial streams (number of spatial streams x) and Y a column vector of received symbols on one subcarrier with entries for each receive antenna (number of antennas y), the CSI is a representation of the channel matrix H (size y rows and x columns). The relation is Y = H * X + noise. H describes the amplitude and phase changes introduced by the channel on each subcarrier. Hence, it is simply a factor in a complex multiplication and does not have any unit. Be aware that H does not only describe the physical channel between the antennas, but also covers the whole signal processing on transmitter and receiver sides.

zeroby0, 2020-09-09:
CSI is a ratio, so it has no units or dimensions. People might write ratios like this in dB, but dB is just a convenient representation of the ratio, not a unit.

zmsenqn, 2020-09-09:
Thank you for your answers.

Stay safe out there.

closed 2020-09-09 by zmsenqn