seemoo-lab/nexmon_csi #114: The amplitude of CSI is particularly small under certain subcarrier

Cyrbaby, 2020-09-14:
greetings,I use Asus RT-AC86U as receiver to extract CSI in a 80MHz bandwidth channel，the scene is a meeting room,but I found out in a subcarrier the amplitude of CSI is very small. May I ask the reason?
[image]

matthiasseemoo, 2020-09-14:
It depends on your environment. If you have many reflections it can happen that the received signal cancels itself out on some subcarriers. Just assume that you have two pathes your signal can travel. On one path you receive a sine signal without phase offset on the other path you receive a 180 degrees phase shifted version of the same signal, namely a -sine. If they are received with the same amplitude, the two signals received over the two different paths will cancel each other. Just try to move your receiver or transmitter to different locations and you shall observe that the channel measurement will be different.

Cyrbaby, 2020-09-18:
> It depends on your environment. If you have many reflections it can happen that the received signal cancels itself out on some subcarriers. Just assume that you have two pathes your signal can [...]

Thanks for the answer！I have misunderstood some concepts.

closed 2020-09-19 by Cyrbaby