seemoo-lab/nexmon_csi #115: Representation of (Guard) Subcarriers at 40/80MHz

Gi-z, 2020-09-14:
Apologies if an issue is not the best way to ask these questions. This seems to be a very active forum on a specific area.

I've seen a few responses refer to guard and pilot subcarriers, which should be removed since they aren't used to carry modulated data (and so wouldn't contain meaningful CSI?). These responses usually point towards the [802.11ac Survival Guide Chapter 2](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html) Table 2-1 which indicates the available subcarrier ranges for each frequency band. I'm somewhat confused as to how some of these numbers line up with the output data from nexmon_csi on the Raspberry Pi 4 (bcm43455c0).

In my examples, I've been collecting CSI using a Pi 4 connected to a 5GHz wireless network operating on channel 36 at either 40MHz or 80Mhz (Sky Router ER115). 

When parsing the CSI payloads, I can see a difference in the number of available subcarriers when operating on the different frequency bands: 128 at 40Mhz, 256 at 80MHz. This is somewhat similar to the numbers seen in Table 2-1 however there appear to be more subcarriers than the table indicates there should be (114 vs 128, 242 vs 256). Am I interpreting the data incorrectly?

The reported pilot subcarriers for 80MHz are at +/- 11, 39, 75, 108. Converting this to 0-255 scale, then there should be pilots at indices 25, 33, 89, 117, 139, 167, 203 and 231. After reading an 80MHz sample capture and plotting it as a heatmap below, I can see differences in CSI at indices which should be used as pilot subcarriers? I am unsure if I am interpreting these values correctly, or if my network is operating in the way I think it is.

(I'm not very experienced with RF so I may be using some terminology incorrectly (I appreciate any corrections)).

[image]

zeroby0, 2020-09-14:
At 40 MHz, the actual sub carriers are `-64 to 0 to +63`. But the table doesn't mention null sub carriers, so it's 114 instead of 128. So for 40 MHz, null sub carriers are: `[-64, -63, -62, -61, -60, -59, -1, 0, 1, 59, 60, 61, 62, 63]`, and pilot sub carriers are:  `[11, 25, 53, -11, -25, -53]`.  I remove both in pre-processing my CSI.

I see a lot of users asking this question, so it's useful to have it here. Thank you for posting this. Btw, did you use the latest nexmon_csi and nexmon for data collection with Pi 4? Mine was unstable on kernel 5.4 for some reason.

Gi-z, 2020-09-14:
Thank you, I wasn't aware of the existence of null subcarriers, which definitely clears up my first point of confusion.

I'm still quite confused as to how the subcarriers as viewed in the data actually align with these indices, as I appear to have CSI data in subcarriers that should either be null or pilots. For example, in this image below I am retrieving packets sent on channel 36 at 40MHz. I am concerned the router I am using may not actually be using an 80MHz channel, as the only options available are 36/80MHz, 36/40Mhz or 44/40MHz. This may be affecting my data as I do believe 36 and 44 are 20MHz channels?

In the image, invariant signals can be observed at subcarrier indices 63, 62 and to a slightly lesser extent 61. However I can see what looks like valid CSI between 3-60. Is this actually valid data in there? There is another cluster of invariant signals between +2 and -4.

[image]

I'm mostly just trying to remove these subcarriers which are not of use to reduce my feature set for deep learning, however I don't want to excessively cut off more than is necessary.

I'm using the version from #46 as I was previously having issues with trapping that this version resolved. Was planning on updating to this newer version however if there are any stability issues I will hold off as I have already struggled to assemble a consistent CSI capture setup!

zeroby0, 2020-09-14:
Your interpretation of the indices is correct. I have not dealt with 80 MHz data yet, but here is what I do with 40 MHz.
```
            nullsubcarriers = [x+64 for x in [-64, -63, -62, -61, -60, -59, -1, 0, 1, 59, 60, 61, 62, 63]]
            csi = csi.drop(columns=nullsubcarriers)

            pilotsubcarriers = [x+64 for x in [11, 25, 53, -11, -25, -53]]
            csi = csi.drop(columns=pilotsubcarriers)
```

Your router is probably set to EU region, so it shows only those 3 channels. You maybe able to get other channels if you set it to US and the firmware supports it. But you should still be fine if it's set to 36/80.

This table has a list of WLan channels: https://en.wikipedia.org/wiki/List_of_WLAN_channels.
[image]

Even though it shows 36 as a 20 MHz channel, using 36/80 in makecsiparams, Nexmon will capture CSI for the entire 80 MHz band as you expect. 40/80, 44/80, 48/80 capture the same 80 MHz channel as 36/80 as well.

In the second image you posted, it does seem like it's only 40 MHz, but that's not because of your router settings. Even when your router is set to 80 MHz, it can still opt to transmit using only 20 MHz or 40 MHz. You'll be able to tell better if you plot a single CSI sample instead, and remove the null sub carriers.

One way to make sure you receive 80 MHz signals is to inject packets using Nexmon from a supported router or another Raspberry Pi (Which I don't really know how to do). But it seems just pinging your router with a laptop over WiFi saturates the 80 MHz as well.

In summary: Your understanding of indices is correct. You should remove both Null and Pilot sub carriers. To saturate 80 MHz, ping the router with your laptop and collect CSI for only the ping reply packets from your router (`makecsiparams -c 36/80 -C 1 -N 1 -m <router Mac> -b 0x88`,  Ping packets start with 0x88)

Additionally, you should undo the effect of AGC before you train your model. AGC causes CSI to vary even when your device is stationary.  You can use RSSI from #46 to undo AGC, and then remove subcarriers.

[image]

Gi-z, 2020-09-15:
This is excellent, thanks. I really appreciate this level of detail and the distinction between the router being set to operate at 80MHz but the devices connected sending 20/40MHz frames is very helpful as I didn't know this was possible. This definitely helps with a few major issues.

closed 2020-09-15 by Gi-z

matthiasseemoo, 2020-09-15:
https://www.cambridge.org/de/academic/subjects/engineering/wireless-communications/next-generation-wireless-lans-80211n-and-80211ac-2nd-edition?format=HB&isbn=9781107016767 <https://www.cambridge.org/de/academic/subjects/engineering/wireless-communications/next-generation-wireless-lans-80211n-and-80211ac-2nd-edition?format=HB&isbn=9781107016767>

This is a very good book showing you how Wi-Fi works on the physical layer.

baldevbarrsiwal, 2020-11-04:
@zeroby0  First of all, thanks a lot Arvind for the detailed explanation,  I have seen there are many facing similar issues. As you have suggest to write a guide #88 and I can volunteer myself for writing it up for this issue

I am still facing a problem to drop the  nullsubcarriers and pilotsubcarriers. Would it be possible to share the link to your Matlab code to get some help. Could you please share your email address if possible?
Thanks a lot for the support.

zeroby0, 2020-11-04:
Hey Baldev! 

Thanks! I think a lot of people would benefit from a guide. 

All of my code is in Python. I get the CSI data into a pandas dataframe, and drop the columns I don't need. I have attached the relevant portion of my code below.  I don't remember how to write Matlab anymore, so can't help you with that XD. Although, I suppose you could take processed CSI samples, dump them into a binary file, and read them from Matlab.

My email is aravind.reddy@iiitb.org.

```Python
'''
pcap.py
====

Provides fast methods to read bcm4355c0 format 
CSI data from .pcap files.
Author: Aravind Reddy V <aravind.reddy@iiitb.org>
License: MIT
'''

__all__ = [
    'read_csi'
]

import os
import numpy as np
import pandas as pd

def _read_csi_next(pcapfile, csi_size):
    """
    Note: Designed for internal use only.
    
    Parameters
    ----------
        pcapfile : File Object
        csi_size : Expected length of CSI in bytes. NFFT * 4
    """

    # Read Frame Size
    pcapfile.seek(8, os.SEEK_CUR)
    frame_size = int.from_bytes(
        pcapfile.read(4),
        byteorder = 'little',
        signed = False
    )

    # Skip some stuff
    pcapfile.seek(56, os.SEEK_CUR)

    # Read CSI data
    pcapfile.seek(8, os.SEEK_CUR)
    csi = np.frombuffer(
        pcapfile.read(csi_size), 
        dtype = np.int16,
        count = int(csi_size / 2)
    )

    # Skip any zero-padding
    pcapfile.seek((frame_size - csi_size - 60), os.SEEK_CUR)

    return csi

def read_csi(pcap_file_path):
    """
    Read CSI data from PCAP file.
    Supports only 40MHz bandwidth,
    and only one Mac ID. You have
    to remove null subcarriers
    yourself.

    Parameters
    ----------
        pcap_file_path : str
    """

    bandwidth = 40

    NFFT = int(bandwidth * 3.2) # Number of channels in FFT
    chunksize = 1024

    csi  = np.zeros((chunksize, NFFT * 2), dtype = 'int16')

    with open(pcap_file_path, 'rb') as pcapfile:
        filesize = os.stat(pcap_file_path).st_size
        pcapfile.seek(24, os.SEEK_SET)

        npackets = 0
        while pcapfile.tell() < filesize:
            if not (npackets % chunksize):
                csi = np.vstack((csi, np.zeros((chunksize, NFFT * 2), dtype = 'int16')))

            csi[npackets] = _read_csi_next(pcapfile, NFFT * 4)
            
            npackets += 1

    # Convert CSI complex numbers to Magnitude.
    csi_converted = np.abs(
        np.fft.fftshift(csi[:npackets, ::2] + 1.j * csi[:npackets, 1::2], axes=(1,))
    )

    return csi_converted

if __name__ == '__main__':
     csi = pd.DataFrame(read_csi('files/csi.pcap'))

     # 80 MHz
     # nullsubcarriers  = np.array([x+128 for x in [-128, -127, -126, -125, -124, -123, -1, 0, 1, 123, 124, 125, 126, 127]])
     # pilotsubcarriers = np.array([x+128 for x in [-103, -75, -39, -11, 11, 39, 75, 103]])

     # 40 MHz
     nullsubcarriers = [x+64 for x in [-64, -63, -62, -61, -60, -59, -1, 0, 1, 59, 60, 61, 62, 63]]
     pilotsubcarriers = [x+64 for x in [11, 25, 53, -11, -25, -53]]

     csi = csi.drop(columns=nullsubcarriers)
     csi = csi.drop(columns=pilotsubcarriers)
     
     # I have not tested this on 80 MHz data yet. It should work, but I can't guarantee it.

```

baldevbarrsiwal, 2020-11-05:
@zeroby0  Thanks Arvind for the support.

I have figured out to how remove the pilot and Null carriers from the CSIdata.

However, by removing the null carriers  `nullsubcarriers = [x+64 for x in [-64, -63, -62, -61, -60, -59, -1, 0, 1, 59, 60, 61, 62, 63]]` and Pilotcarriers `  pilotsubcarriers = [x+64 for x in [11, 25, 53, -11, -25, -53]]` does not completely remove the static data from CSI buffer 

On x-Axis do you take `x= -(nfft/2):1:(nfft/2-1); `     or  `x= -(nfft/2 -1 ):1:(nfft/2); `   ?? Because this may result in removing the wrong carriers from data.

Here is my Matlab code snip

`%% plot

choice = input('Enter a number: ');
switch choice
    
% PilotSubcarriers = [12 40 54 76 90 118] ie = 64 ( ±11, ±25, ±53 ) where x-axis ends with -(nfft/2) == -64 in file "plotcsi.m" x = -(nfft/2):1:(nfft/2-1); 

% NullSubcarriers = [1 2 3 4 5 6 64 65 66 124 125 126 127 128] ie = 64 ([-64, -63, -62, -61, -60, -59, -1, 0, 1, 59, 60, 61, 62, 63] ) where x-axis ends with -(nfft/2) == -64

% csi_buff - Unchanged_CSI_Columns = [1 2 62 63 64 65 66 67 68 128] % these coulms have constant unchanged value
%% Pilot and Null both removed as per #115
    case 1
    PilotNullsubcarriers = [ 1 2 3 4 5 6 12 40 54 64 65 66 76 90 118 124 125 126 127 128 ] %% Pilot and Null both removed as per #115
    csi_buff(:,PilotNullsubcarriers) = [] %% Remove the listed coloumn of the list "PilotNullsubcarriers"
    NFFT = NFFT-numel(PilotNullsubcarriers)
    plotcsi(csi_buff, NFFT, false)
    
   case 2
 %%   Pilot and Unchanged carrier are removed  " csi_buff - Unchanged_CSI_Columns" ##################

    PilotUnchangedSubcarriers =  [ 1 2 12 40 54 62 63 64 65 66 67 68 76 90 118 128 ] %% Pilot and columnIndicesToDelete both removed
    csi_buff(:,PilotUnchangedSubcarriers) = [] %% Remove the listed coloumn of the list "Pilotand UnchangedSubcarriers"
    NFFT = NFFT-numel(PilotUnchangedSubcarriers)
    plotcsi(csi_buff, NFFT, false)
    
    case 3
        % without remooving the null and pilot subcarriers
    plotcsi(csi_buff, NFFT, false)
    
    otherwise
    warning('Unexpected plot type. No plot created.')
end

`

**How did you identified the Null Carriers?**
Thanks again for the contribution

zeroby0, 2020-11-05:
@baldevbarrsiwal 

It goes from `(-nfft/2):0:(nfft/2-1)`. Your indices look correct though.

> does not completely remove the static data from CSI buffer

Didn't understand this. If you mean the CSI looks same from one frame to another, that is to be expected. Especially at the edges, because wifi applies an inverted U shaped spectral mask, which would reduce dynamic contrast at the edges.

I get the indices for null carriers from here: https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html

baldevbarrsiwal, 2020-11-10:
@zeroby0 Thanks Arvind for the hints, I managed to remove the outliers and make the CSIdata more presentable.

You have mentioned for AGC removal but I couldn't get how to remove it in #46 

I'm using the your repository installation procedure with Kernal version 4.19.
https://github.com/zeroby0/nexmon_csi/tree/release-pi-buster-4.19.97-plus/releases/pi3Bplus-buster-4.19.97

Could you please point me how to remove AGC affect.

Thank you.

satya2891, 2023-08-12:
> Your interpretation of the indices is correct. I have not dealt with 80 MHz data yet, but here is what I do with 40 MHz.
> 
> ```
>             nullsubcarriers = [x+64 for x in [-64, -63, -62, -61, -60, -59, -1, 0, 1, 59, 60, 61, 62, 63]]
>             csi = csi.drop(columns=nullsubcarriers)
> 
>             pilotsubcarriers = [x+64 for x in [11, 25, 53, -11, -25, -53]]
>             csi = csi.drop(columns=pilotsubcarriers)
> ```
> 
> Your router is probably set to EU region, so it shows only those 3 channels. You maybe able to get other channels if you set it to US and the firmware supports it. But you should still be fine if it's set to 36/80.
> 
> This table has a list of WLan channels: https://en.wikipedia.org/wiki/List_of_WLAN_channels. [image]
> 
> Even though it shows 36 as a 20 MHz channel, using 36/80 in makecsiparams, Nexmon will capture CSI for the entire 80 MHz band as you expect. 40/80, 44/80, 48/80 capture the same 80 MHz channel as 36/80 as well.
> 
> In the second image you posted, it does seem like it's only 40 MHz, but that's not because of your router settings. Even when your router is set to 80 MHz, it can still opt to transmit using only 20 MHz or 40 MHz. You'll be able to tell better if you plot a single CSI sample instead, and remove the null sub carriers.
> 
> One way to make sure you receive 80 MHz signals is to inject packets using Nexmon from a supported router or another Raspberry Pi (Which I don't really know how to do). But it seems just pinging your router with a laptop over WiFi saturates the 80 MHz as well.
> 
> In summary: Your understanding of indices is correct. You should remove both Null and Pilot sub carriers. To saturate 80 MHz, ping the router with your laptop and collect CSI for only the ping reply packets from your router (`makecsiparams -c 36/80 -C 1 -N 1 -m <router Mac> -b 0x88`, Ping packets start with 0x88)
> 
> Additionally, you should undo the effect of AGC before you train your model. AGC causes CSI to vary even when your device is stationary. You can use RSSI from #46 to undo AGC, and then remove subcarriers. Here is an example of AGC messing with CSI.
> 
> [image]

Hi,

In this command, (`makecsiparams -c 36/80 -C 1 -N 1 -m <router Mac> -b 0x88`, Ping packets start with 0x88)
Even after specifying the router mac ID and pinging the packets from Linux machine why it is not collecting?