seemoo-lab/nexmon_csi #117: The phase change from subcarrier 66 to subcarrier 64 is always abnormal

Cyrbaby, 2020-09-19:
greetings,I use Asus RT-AC86U as receiver to extract CSI in a 80MHz bandwidth channel，no matter the scene is indoor or outdoor，the phase change from subcarrier 66 to subcarrier 64 is always abnormal，different from other phase changes，should I deal with this phase offset？
[image]
[image]
[image]

zel602, 2020-09-28:
I have meet the same question，is the subcarrier -65 and -64 inconsistent?I have also analyze the  example capture file utils/matlab/example.pcap ，and found the same question. Anyone who meet this before?@matthiasseemoo @jlinktu
[image]

jlinktu, 2020-09-29:
For 80MHz 802.11 VHT frames the tones of the three upper 20MHz subbands (subcarriers -64 and above) of i. a. the LTF are multiplied by -1 (rotated by 180 degree), probably producing the phase jump in the channel estimation between subcarriers -65 and -64. (see 802.11-2016 21.3.7.5 Definition of tone rotation)