seemoo-lab/nexmon_csi #119: How to get CSI for only Ping requests?

zeroby0, 2020-09-25:
How do I collect CSI only for a specific type of packet? I would like to collect only for ping to ensure that all my packets have a full 80 MHz bandwidth. I cannot use packet injection because I would like the frames to be coming from any generic router.

`makecsiparams`'s -b flight might be useful here, but I cannot find the exact number to use. I tried to loop from 0x00 to 0xff and see which corresponds with the number of packets ping sends, but the CSI collection stops after changing the number a few times.

matthiasseemoo, 2020-09-27:
You need to extend both the arm firmware patch as well as the ucode to filter for bytes indicating ping packets. By writing to the shared memory (shm), you can pass information from the arm firmware to the ucode (as it is done in [1]). The information is evaluated in the ucode to filter frames (as it is done in [2]).

[1] https://github.com/seemoo-lab/nexmon_csi/blob/194d7396d907ccfddae65ff95e7f464648cf6c7c/src/ioctl.c#L162
[2] https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/csi.ucode.bcm43455c0.7_45_189.patch#L122

zeroby0, 2020-10-14:
@matthiasseemoo Matthias, how do I access arbitrary bytes in the frame from ucode? 

I used `and     [18,off1], 0xff, [4,off5]` like in https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/7e3f9f720e1eb12ef11afd855515981d5a3b715b of #46 , so I should get the first byte in Frame's payload, but CSI collection just stops.

Also, from the CSI paper, the WLAN header is replaced with Ethernet header, so I don't know the index of the byte I should be looking at either. With Ethernet header, I would look at IP.Protocol byte at `29 & 0xFF00` to see if it's ICMP.

SimoneDelPrete, 2024-02-08:
> @matthiasseemoo Matthias, how do I access arbitrary bytes in the frame from ucode?
> 
> I used `and [18,off1], 0xff, [4,off5]` like in [7e3f9f7](https://github.com/seemoo-lab/nexmon_csi/commit/7e3f9f720e1eb12ef11afd855515981d5a3b715b) of #46 , so I should get the first byte in Frame's payload, but CSI collection just stops.
> 
> Also, from the CSI paper, the WLAN header is replaced with Ethernet header, so I don't know the index of the byte I should be looking at either. With Ethernet header, I would look at IP.Protocol byte at `29 & 0xFF00` to see if it's ICMP.

Hi, by any chance, have you solved this issue?