seemoo-lab/nexmon_csi #120: RFC: Releases

zeroby0, 2020-09-27:
# Motivation

Nexmon_CSI is in active development. As such, code cloned from this repository may not work for some combinations of Hardware, OS, and Nexmon_CSI version (commit).

Providing a known good configuration saves the user a substantial amount of time. This RFC describes a method to create _releases_ which do exactly that. Please write your comments and suggestions to streamline this process.

# Design
Releases are titled in this format: `Device-OS-Identifier`.
For example, `Pi3Bplus-Buster-4.19.97` is a release targeted for Raspberry Pi 3B+ running Raspbian Buster with Kernel v4.19.97.

## Modifications

A new directory called `releases` is added to the current repository. A template release directory is added to help contributors.

## Creating a new release

To create a new Release:
1. A new Issue is created with `Release <Release Title>`. Discussion about the Release happens here.
2. A new Pull Request is made titled ` Release<Release Title>. Fixes #<issuenumber>`

The Pull Request should create a new directory with the same name as the Release to the `releases` directory.
This new directory should have a Readme file describing the release and installation instructions, and optionally any scripts to automate the installation process.

Releases are deterministic. Installing a release on a new device should *always* produce the same result. The release should be tied to a commit to ensure that.

See https://github.com/zeroby0/nexmon_csi/releases/tag/v0.1.1 for an example Release document.

matthiasseemoo, 2020-09-27:
Hi zeroby0,
thank you for your release strategy suggestion.
However, I would not want to add an additional directory for releases. First of all, Both RPi3 and 4 have exactly the same Wi-Fi chip. And if you are running the same OS release, the CSI extractor shall work the same. Additionally, we generally keep multiple versions of patched drivers. So that even older OS releases can be used with the latest state of the repository.

For handling versions, we (at least intend to) use git tags. Those tags will also end up in the built firmware (see https://github.com/seemoo-lab/nexmon_csi/blob/master/src/version.c). Then, you always know exactly on which repository state a firmware was built. However, I have to admit, that lately we did not really use tags anymore.

@jlinktu: I think we should start adding tags to working states of our releases. Preferably using a versioning scheme such as https://semver.org/. Before tagging a state of the repo, we shall execute regression tests on all supported platforms so that we always ensure that each release works as expected.

zeroby0, 2020-10-05:
Hi Mathias!

Sorry for the delayed response. Well labelled git tags would be equally good! Could you enable Wiki as well? There's hordes of valuable info in the issues, but it's not well organized.

matthiasseemoo, 2020-10-05:
Done ;-)

zeroby0, 2020-10-06:
Awesome! It looks like wiki editing is restricted to collaborators only, please [change it to public](https://docs.github.com/en/free-pro-team@latest/github/building-a-strong-community/changing-access-permissions-for-wikis) so all of us can contribute. Thanks :)

matthiasseemoo, 2020-10-07:
I do not want to open the wiki for public editing as it makes it harder to review and approve changes. You could clone the wiki locally and send us your proposed changes as a patch.

zeroby0, 2020-10-08:
I didn't know that was possible. I'll do that. Thank you.

closed 2020-10-08 by zeroby0