seemoo-lab/nexmon_csi #123: ASUS with nexmon frequently crash and reboot when run tcpdump

yujianyuanhaha, 2020-09-27:
hi,

thanks for your amazing teamwork of nexmon, I have been use the tool for a while and get the CSI I want. However, it is annoying to me the **router (model ASUS RT-AC86U) will frequently restart when I run tcpdump to collect CSI for longer time**. Each time the router reboot I have to flash in nexmon again. It is bit random how long the tcpdump will last, usually cannot last longer than 10min.

Several point I checked

1. close wifi and other network connection, the host machine is linked to sniffer router only.

2. tcpdump for shorter time. The router has near 50M internal storage, and can last at most 80 seconds. However, it happen still sometime. If we choose  to save `.pcap` file in external storage `/mnt/sda1/` (a USB flash driver), it usually crash down within 10minutes. There is indeed few time it can last as long as 30 miniutes and I got over 1G `.pcap` files.

This bug has limit my application of long-time data collecting more than 10min.

Anyone meet the issue before and have a solution.

Best.

zel602, 2020-09-28:
I meet this issue before，if I I have a period of time between twice tcpdump ,I will get 0 packets all the time,I have to reboot  ASUS RT-AC86U,so I keep running tcpdump even if I am moving the position of the transmitter, this may mitigate this problem.
By the way,do you meet the phenomenon that the CSI's subcarrier -65 and subcarrier -64 are inconsistent ,when using the 80MHz channel.

yujianyuanhaha, 2020-09-28:
@zel602 My problem is `ssh` break down, not get 0 packet.

For you issue, you may need to open two terminal, in one terminal run `tcpdump`, while it is stuck in another terminal switch off-n-on interface by `wl radio off; wl radio on; ifconfig eth6 off; ifconfig eth6 on`, then you can see tcpdump resume collection.

it make sense subcarrier inconsistent **since 80MHz bandwidth is compose of 4 20MHz bandwidth, besides null/ guard subcarrier, there should always be sudden inconsistent at each 1/4 point (i.e. -64 0, +64)**

zel602, 2020-09-28:
> @zel602 My problem is `ssh` break down, not get 0 packet.
>
> For you issue, you may need to open two terminal, in one terminal run `tcpdump`, while it is stuck in another terminal switch off-n-on [...]

Thanks for your answer!You have solved my another question!
For your issue,I think you can change this parameter in router setting,which may be helpful.
[image]
By the way,for the subcarrier inconsistent problem,what makes me feel confused is  the CSI's subcarrier -65 and subcarrier -64 are inconsistent, however,the phase of subcarrier 63 and subcarrier 64 are consistent.Do you meet same phenomenon before?How much bandwith between  subcarrier -65 and subcarrier -64 ?
[image]
CanI get your contact information?My Email is 419046121@qq.com.Maybe we can have more communication.

yujianyuanhaha, 2020-09-28:
hi @zel602 

i set disable idle timeout very early, it should be other reason, I feel like it maybe memory overflow.

Again, since there is a gap between each 20M bandwidth, it is not surprise there is sudden change/ inconsistent between -65,-64. I am not clear of the size of gap as well.

Feel free to post your question right here, this is the power of open source and Github.

zel602, 2020-09-29:
> hi @zel602
> 
> I disable idle timeout very early, it should be other reason, I feel like it maybe memory overflow.
> 
> Again, since there is a gap between each 20M bandwidth, it is not surprise there is sudden change/ inconsistent between -65,-64. I am not clear of the size of gap as well. A easier way is drop few subcarriers since CSI is pretty rich.
> 
> Feel free to post your question right here, this is the power of open source and Github.

Hi!

I never meet the situation ssh break down.May the firmware  you use is the matter.

Could you share the phase figure for your measured CSI in 80MHz?I think we can get some results by comparing the phase figure.

yujianyuanhaha, 2020-10-02:
several reason I feel like it may ease the crash down:
1. change the router Ip from 192.168.0.x to 192.168.2.x
2. save into smaller .pcap file 
3. load the nexmon tool via `/sbin/rmmod dhd;  /sbin/insmod /jffs/dhd.ko` directly in the router

tweigel-dev, 2020-12-02:
sry for the trivial question but how you install tcpdump at the Asus, because my asus isn't conneted to the internet. To copy the tcp bin does not work for me.  @yujianyuanhaha  and @zel602

zeroby0, 2020-12-02:
Install TCPdump
===============
All the below commands should be run on your computer; as signified by the `user@vm$`. The compile toolchain and environment is setup in 8th and 9th steps of https://github.com/seemoo-lab/nexmon_csi#bcm4366c0. You would have run those steps while installing nexutil. Run these commands in the same terminal, or run 9th command before running these scripts in a new terminal.

- Get Merlin Firmware
  - `user@vm$ git clone https://github.com/RMerl/asuswrt-merlin.ng.git`

- Cross compile LibPCAP. Tcpdump uses it under the hood.
  - `user@vm$ cd asuswrt-merlin.ng/release/src/router/libpcap`
  - `user@vm$ CC=aarch64-linux-gcc AR=aarch64-linux-ar RANLIB=aarch64-linux-ranlib AS=aarch64-linux-as LD=aarch64-linux-ld NM=aarch64-linux-nm ./configure --host=x86_64-unknown-linux-gnu --with-pcap=linux`
  - `user@vm$ make`

- Cross compile TCPdump
  - `user@vm$ cd ../tcpdump-4.4.0`
  - `user@vm$ CC=aarch64-linux-gcc AR=aarch64-linux-ar RANLIB=aarch64-linux-ranlib AS=aarch64-linux-as LD=aarch64-linux-ld NM=aarch64-linux-nm ac_cv_linux_vers=2 CFLAGS=-static CPPFLAGS=-static LDFLAGS=-static ./configure --host=x86_64-unknown-linux-gnu`
  - `user@vm$ make`

- Copy TCPdump to router and make it executable
  - `user@vm$ scp tcpdump admin@<address of your rt-ac86u>:/jffs/tcpdump`
  - `user@vm$ ssh admin@<address of your rt-ac86u> "/bin/chmod +x /jffs/tcpdump"`

tweigel-dev, 2020-12-02:
Thanks for the tutorial.
but one step failed while Cross compile LibPCAP:
```
 CC=aarch64-linux-gcc AR=aarch64-linux-ar RANLIB=aarch64-linux-ranlib AS=aarch64-linux-as LD=aarch64-linux-ld NM=aarch64-linux-nm ./configure --host=x86_64-unknown-linux-gnu --with-pcap=linux
configure: WARNING: If you wanted to set the --build type, don't use --host.
    If a cross compiler is detected then cross compile mode will be used.
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for x86_64-unknown-linux-gnu-gcc... aarch64-linux-gcc
checking for C compiler default output file name... 
configure: error: C compiler cannot create executables
See `config.log' for more details.
```
@zeroby0  Did you have a clue whats going on ? 
Also which makefile you mean to run make on ?

zeroby0, 2020-12-02:
I saw that error yesterday in one of the issues. Have you installed g++ and ran step 8 and 9?

crownz-sec, 2021-12-22:
> several reason I feel like it may ease the crash down:
> 
> 1. change the router Ip from 192.168.0.x to 192.168.2.x
> 2. save into smaller .pcap file
> 3. load the nexmon tool via `/sbin/rmmod dhd;  /sbin/insmod /jffs/dhd.ko` directly in the router

Hi, have you solved the reboot problem yet? I found the same reboot issue when using [UbiLoate](https://github.com/IMDEANetworksWNG/UbiLocate) based on Nexmon. So, I came to Nexmon project to find the solution. I'll try what you said, but I still want to ask if there are other ways, and ultimately how long can you keep the router from rebooting? I'm sorry that I'm responding to the message from a long time ago, but this issue really bothers me so much that I have to do this.