seemoo-lab/nexmon_csi #125: Access to AGC values

salomoneli, 2020-10-05:
Hi, 
is it somehow possible to get access to the Automatic Gain Control (AGC) values on the Raspberry Pi B 4?

Gi-z, 2020-10-05:
Use this version of nexmon_csi from #46, which inclues RSSI and Frame Control in the frame format just after the magic bytes (1111).

zeroby0, 2020-10-05:
I saw it discussed in the issues a while ago. I'm not going to search for them, but I think there were at least 2 issues which tried to do that.

salomoneli, 2020-10-05:
I think the issues that you are referring to are the ones where it is described that RSSI should be used to cancel out the influence of the AGC (e.g. this one [here](https://github.com/seemoo-lab/nexmon_csi/issues/93)).

But my question is now, if and how the AGC value could be accessed. I haven't found any clear answer to this question so far.
The Linux 802.11n CSI Tool for example provides the AGC value (see: [FAQ, Question 2 - "B. Inspecting the CSI"](https://dhalperi.github.io/linux-80211n-csitool/faq.html)).

zeroby0, 2020-10-07:
Just a speculation at this point, but the hardware header _may_ contain the agc value:

https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/src/csi_extractor.c#L105

T3rO, 2020-10-21:
I extracted the RxChan. The first byte is the channel and the other byte is some value that didn't change during my measurements. It is different for different channels but probably not agc

T3rO, 2020-10-25:
According to this file: https://github.com/MerlinRdev/86u-merlin/blob/master/release/src-rt-5.02hnd/bcmdrivers/broadcom/net/wl/impl51/4365/src/include/d11.h#L2971 PhyRxStatus_2 contains "lnagn", "pgagn", and "foff". I don't know for sure what these variables mean, but maybe "low noise amplifier gain" etc.
In one of the nexmon paper there is an image showing the components of the Wifi chip and some of the amplifiers are named lna and pga. 

I also don't know how the agc works, but if it controls these amplifiers, maybe these values are useful.

I noticed that they influence the rssi (chanspec 36/20, transmitter and reciever were close next to each other. I couldn't reproduc ethe result, when they were further apart):
[image]

tweigel-dev, 2020-12-02:
@T3rO could your share your method of modifying nexmon_csi to gather more then CSI with us? Because this values are very interesting. It would be great if we could extend  nexmon_csi (here or in a fork) to gather nearly the same information then Linux_csi by @halperin 
With the spirit of open source ;-)

T3rO, 2020-12-02:
Take a look at the fork from mzakharo and the csi_extractor.c file. In this fork the RSSI and Phystatus are recovered. I just plotted them in the above answer. However I think I misinterpreted the PhyRxStatus2 value. It is just the rssi. For other phy types or firmware versions the meaning of this value might be different.

In order to extract other values I suggest appending variables to the csi_udp_frame struct and setting them in the create_new_csi_frame function. See how RSSI is extracted for an example.

tweigel-dev, 2020-12-02:
@T3rO Tanks!  Did you think such information like snr or noise are accessible ?

T3rO, 2020-12-02:
I don't know if these values are included in the d11rxhdr, but basically you can also read all the phyreg values. Maybe they are included there or can be calculated from some of these phyreg values.

So far I managed to read some gain values from the chip's gain stages. This file describes the phyreg: https://github.com/MerlinRdev/86u-merlin/blob/master/release/src-rt-5.02hnd/bcmdrivers/broadcom/net/wl/impl51/4365/src/wl/phy/wlc_phyreg_ac.h

CDeLeon94, 2021-09-24:
Was any progress ever made in getting the AGC values?
Just stumbling upon this issue now, trying to figure out how to deal with it.

T3rO, 2021-09-25:
Yes, I think I figured it out to some degree. 
Basically, you have to read two PHY-Register entries: GainInfo and auxphystats. After applying some bitshifts and bitmask you will get a gaincode A and gaincode B. These two gaincodes contain gaincodes for all the gainstages except TrLoss. TrLoss has its own phyreg entry I think.
To get the actual gain values you can use the gaincodes and look up the respective gain values in a PHY-Table called Gaintable. I think some gain values can not be found in the gaintable, but sometimes the gainvalue is supposedly the gain code multiplied by 3 or just the gaincode itself.

CDeLeon94, 2021-09-28:
I appreciate the info, so, I'm assuming I need that file you referenced above that describes the phyreg. 
Do you happen to have a copy of that?
It's been removed from github and I can't seem to find it via google.

CDeLeon94, 2021-09-28:
https://github.com/seemoo-lab/nexmon_csi/pull/46#issuecomment-929382492

new reference for phy registers, naming seems to be different.

T3rO, 2021-09-28:
I published my csi/gain recovery patch here: https://github.com/T3rO/nexmon_csi_gain. Hopefully this can help. If anything is unclear, feel free to ask.

T3rO, 2021-09-28:
> [#46 (comment)](https://github.com/seemoo-lab/nexmon_csi/pull/46#issuecomment-929382492)
> 
> new reference for phy registers, naming seems to be different.

That file is not for the phy register but for d11. You need wlc_phyreg_ac.h. But I don't know where you can find it.