seemoo-lab/nexmon_csi #126: get RSSI from ASUS RT_AC86U router in monitor mode

yujianyuanhaha, 2020-10-05:
greetings,

How can we get the RSSI on the ASUS router running nexmon tool.

I know there is a command `wl rssi` and `wl noise` that is straight forward. Hoever, it only work for router in regualr mode where one router is communication with the other.
While in our nexmon router is in monitor mode and do sniffering, which mean it does not **connect** to another router, so the command mention would not work ...

Is there any idea we can get that? @DanielAW @schmittner @jlinktu 

Thanks

mzakharo, 2020-10-05:
The RSSI patch you mention is not Raspberry PI specific, should work for ASUS too. Have you tried?

yujianyuanhaha, 2020-10-06:
@mzakharo  apparently that patch doesn't work for ASUS

mzakharo, 2020-10-06:
got to elaborate, what do you mean 'does not work'.  the line below is device-agnostic. If it does not work for you, it is a good starting point  for you to debug yourself and to figure out why. 
```c
 wlc_phy_rssi_compute(wlc_hw->band->pi, wlc_rxhdr);
 last_rssi = wlc_rxhdr->rssi;
```

Deltaaaaaaaaaa, 2020-10-07:
@mzakharo 
I'm also trying to get RSSI along with CSI on AC86U, Thankyou for your instruction!
I've tried modifying the csi_extractor.c file, and received rssi value secussfully. 
However, the extracted rssi values are always the same across the 4 attennas. It seems frames from different attennas in one measurement (one packet from tx) are reporting the rssi from one attenna. 
I really wonder what is it like on Raspberry PI, did the same problem occured to you when getting rssi from different attennas?
I noticed in [your commit on RPI](https://github.com/seemoo-lab/nexmon_csi/commit/7e3f9f720e1eb12ef11afd855515981d5a3b715b), both the patch file (src/csi.ucode.bcm43455c0.7_45_189.patch) and the csi_extractor.c file were modified. How does the added line "and	[3,off1], 0xff, [4,off5]" in the patch file work? How should i modify the patch file for AC86U (src/csi.ucode.bcm4366c0.10_10_122_20.patch) to make it report the rssi for each attenna? 
Could you please give us a hint?
Thankyou so much!

mzakharo, 2020-10-07:
`wlc_rxhdr->rssi` is a computed value, derived from all 4 antennas-> the source code for `wlc_phy_rssi_compute`  is available if you google for it.   RPI-4 has only 1 antenna, so this is no problem.  Now as far as getting separate RSSI for each antenna on ASUS, you have to dig those out yourself, they should be in there, part of phybits: https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/b79d935c846fbae6ab9b28eafe8416358671feb3 and https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/a9d2a2fb5d93b555d53fa8cea2756280345103fd . Good luck!

jlinktu, 2020-10-07:
You can use the attached patch to include received power per antenna on the _bcm4366c0_ in the corresponding CSI UDPs.
With the patch the ucode extracts the value of the correct antenna from the phystatus and appends it to the header that is passed to the arm core together with the CSI data. The arm core just receives and converts the value before inserting it as a `int16` in the CSI UDP. Note that this adds another 2 bytes to the header in the CSI UDP before the actual CSI values.
This is roughly what `wlc_phy_rssi_compute` does but without combining the per antenna values. `wlc_phy_rssi_compute` actually might do more like additionally compensating temperature, applying per band offets etc. If the raw receive power extracted by the patch isn't fine enough for you and you are interested you can find the function called by `wlc_phy_rssi_compute` at `0x21c532`.

[add_rxpwr_extraction_bcm4366c0.patch.txt](https://github.com/seemoo-lab/nexmon_csi/files/5341003/add_rxpwr_extraction_bcm4366c0.patch.txt)
Clone `nexmon_csi` as usual and apply with `git apply add_rxpwr_extraction_bcm4366c0.patch.txt` before compiling.

yujianyuanhaha, 2020-10-07:
@jlinktu sounds make sense, thanks a lot

yujianyuanhaha, 2020-10-09:
[data format description](https://github.com/seemoo-lab/nexmon_csi#analyzing-the-csi) well descrip my problem and I find the solution for it.

Thanks

tweigel-dev, 2020-12-02:
@yujianyuanhaha I am writing a python script to extract CSI. 
Currently this is only working with bcm43455c0 but i want to implement an patch for bcm4366c0.
https://github.com/tweigel-dev/nexmon_csi/tree/feature/python 
look at utils/python

zeroby0, 2020-12-02:
@tweigel-dev looks nice!

It would be faster and less memory hungry to load the entries into pandas tables than put them into classes (although I too am guilty of this XD). 

My version of python csi explorer is here: https://github.com/zeroby0/csi-explorer. I implemented it in browser and terminal, feel free to take code from there :)

tweigel-dev, 2020-12-02:
@zeroby0  thx ^^ I think it's great that we do the same things. we should share more to avoid double work ! 👍🏻  
Did you also implemented a subscriber at port 5500 to sniff the output of nexutils with python? I want to do this in near future to pass the csi directly to plotting or some other algorythms.

zeroby0, 2020-12-02:
@tweigel-dev Yess! I have implemented sniffing the port, and have looked into ways for optimising it. The optimised version for reading pcap files is also done. I don't know if we can run python on the router though.

 I'll create a github org and add you?

tweigel-dev, 2020-12-02:
@zeroby0  sure !

Gi-z, 2020-12-02:
@zeroby0 @tweigel-dev Hi, I've got some code doing some similar stuff here: https://github.com/Gi-z/CSIKit, as I've got some implementations I've been working on for sensing with the Pi. If you're pooling work together, I'd love to get involved.

zeroby0, 2020-12-02:
Awesome! I'll add you too!

yujianyuanhaha, 2020-12-20:
@jlinktu  may I know how the **extra 2 byte** I mention is presented for your [solution](https://github.com/seemoo-lab/nexmon_csi/issues/126#issuecomment-704981416)?

closed 2023-12-26 by jlinktu