seemoo-lab/nexmon_csi #128: tcpdump stop after time and wrong chanspc 0x6863

csmibrahim, 2020-10-07:
I followed all the steps and checked the FAQ. I am using Raspberry Pi 3 Model B+ and listing to a router on channel 11/20.
The tool works fine for few minutes/hours then stop. Does not matter if writing to a file or printing on the terminal . I added the channel in regulation based on a suggestion from previous issues. In the beginning, the configuration is as I define using makecsiparams and nexutil, then when tcpdump stop I check with nexutil -k and it gave wrong chanspec 0x6863. When I check 
If I reboot or rebuild it works fine again.  I removed the wpa_supplicant but this does not make any difference. 

I went through the related issues here but nothing fixed the problem. Is there a permanent fix for this problem?

zeroby0, 2020-10-07:
Did you check #46? Add commits 82161c4 and 18339e5 and that might solve the instability.

Or install this known good configuration which includes those fixes: [Release pi-buster-4.19.97-plus](https://github.com/zeroby0/nexmon_csi/blob/release-pi-buster-4.19.97-plus/releases/pi-buster-4.19.97-plus/README.md)

jlinktu, 2020-10-07:
This probably means the chip crashed.

As zeroby0 just mentioned, some fixes by mzakharo (https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/82161c4199d112aedae5fb4ec2d3f191ff387e4c and https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/18339e5c2c7b14296141939b639bbc68aeb5ece8) that are not yet merged solved some crashes in the past.

To locate the origin of a crash it is helpful to provide a crash report, check for it in `dmesg` or `/sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics` (only readable by root) right after the crash happened.

csmibrahim, 2020-10-13:
Thank you @zeroby0 and @jlinktu. 
The workaround to skip AMSDU frames was the fix for this problem as in the shared commits.
Thanks for the tips too.

closed 2020-10-13 by csmibrahim

zeroby0, 2020-10-21:
> solved some crashes in the past.

@jlinktu , are the crashes fixed as of the latest commit? I have not tried running the latest commit yet.

zeroby0, 2020-10-22:
After running the latest commit over night, it stopped with chanspec `0x6863`.

Additionally, WiFi APs on some channels are not scannable with the latest firmware installed, and the wlan0 interface becomes completely un-scannable when the firmware crashes. I was not able to get a forensics report though, `root` stopped working when I ran dmesg.

zeroby0, 2020-10-24:
```
root@raspberrypi:/home/pi/scans# cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics
o_init: host_enab 0
000000.148 Overriding vendor id = 0x14e4
000000.148 Overriding device id = 0x43ab
000000.149 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.171 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.199 wl0: wlc_stf_txcore_shmem_write: No clock
000000.200 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.205 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.206 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.206 TCAM: 256 used: 237 exceed:0
000000.207 reclaim section 1: Returned 118076 bytes to the heap
000000.207 reclaim section 4: Returned 44 bytes to the heap
000000.208 sdpcmd_dpc: Enable
000000.250 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.252 wl0: unable to find iovar "rsdb_mode"
000000.252 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.253 wl0: unable to find iovar "tdls_enable"
000000.253 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000116.679 wl0: invalid chanspec 0xe12a
000116.679 wl0: wlc_iovar_op: chanspec BCME -20 (Bad Channel)
000127.671 wl0: unable to find iovar "toe_ol"
000127.671 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000127.671 wl0: wl_open
000127.683 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000147.964 wl0: unable to find iovar "toe_ol"
000147.965 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000147.967 wl0: unable to find iovar "nd_hostip_clear"
000147.967 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
000166.142 wl0: invalid chanspec 0xe12a
000166.142 wl0: wlc_iovar_op: chanspec BCME -20 (Bad Channel)
000228.607 wl0: invalid chanspec 0xe12a
000228.608 wl0: wlc_iovar_op: chanspec BCME -20 (Bad Channel)
000364.894 wl0: invalid chanspec 0xe12a
root@raspberrypi:/home/pi/scans# nspec BCME -20 (Bad Channel)
```

This is the forensics. The chanspec reported by `nexutil -k` is not what is shown in forensics, and doesn't seem to change when set with nexutil. CSI collection stops at this point.