seemoo-lab/nexmon_csi #131: get RSSI from NEXUS phone in monitor mode

yujianyuanhaha, 2020-10-14:
greetings,

How can we get the RSSI on the NEXUS running nexmon tool?

Like the post [get RSSI on ASUS WiFi](https://github.com/seemoo-lab/nexmon_csi/issues/126) many command only work for router in regular mode where one router is communication with the other. While in our nexmon router is in monitor mode and do sniffering, which mean it does not connect to another router, so the command mention would not work ...

Is there any idea we can get that? @DanielAW @schmittner @jlinktu

Thanks

yangtat, 2021-01-09:
Did you finally extract the RSSI using the Nexus 5?
After installing nexmon on the phone, Even if the monitor mode is 0, I can't connect to WiFi, and I can't collect RSSI

yujianyuanhaha, 2021-01-10:
@yangtat 
nope, I later put another well-functioned phone to collect CSI independently。
You can always go back, disable the monitor mode and connect to a AP using command tool like `iw`

yangtat, 2021-01-11:
@yujianyuanhaha  Thank you very much，lt seems that smartphone cannot get the rssi under monitor mode.