seemoo-lab/nexmon_csi #132: ASUS RT_AC86U CSI captures 0 packets when using tcpdump and throws driver error

hanabhp, 2020-10-14:
Hello,

Thank you the this amazing tool!

I’m trying to collect CSI on ASUS RT_AC86U. I am going to describe the setup and steps I have taken in the following, and would appreciate it if folks who worked on this can verify my steps, and give their feedback.

1. When I installed nexmon_csi, the router SSID disappeared. Reading the previous issues, this seems to be a normal case. Right?
2. To capture CSI, I am using two routers one w/ nexmon_csi and the other runs as a normal router w/o nexmon tool. Both the routers are set to listen to the same channel (I set the channels from router GUI to 157/80), and send ping requests to the router w/o nexmon_csi (Is this a valid way to inject packets?) . However, when running `nexutil -k` to manually set the channel, It throws __nex_driver_io and 0x6863, 85/160 which is an invalid channel.
3. When I first run tcpdump, no errors appear. However, after that  __nex_driver_io appears again. And yet 0 packet is captured.
4. When rebooting the router, I use `/sbin/rmmod dhd; /sbin/insmod /jffs/dhd.ko` to reload nexmon_csi. However, It agains throws __nex_driver_io error. What am I missing?!! @schmittner @jlinktu @yujianyuanhaha 

Thank you,

yujianyuanhaha, 2020-10-15:
hi @hanabhp 
1. it is normal, router in monitor mode may lost SSID
2. (a).  the GUI is not guarantee correct sometimes, double check with command line to see the channel is correct   (b)`ping` cannot guarantee sufficient packets injection, a suggested way is using `iperf3`: you may need a extra mobile phone or laptop as client to iperf3 with your normal mode router (w/o nexmon), then the hacked router (w/ nexmon) as sniffer to tcpdump
3. sometime the router would restart itself, notice the led lights. If the router reboot, load in `/sbin/rmmod dhd; /sbin/insmod /jffs/dhd.ko` and make sure you `ifconfig eth6 up`

You may need to record a video and upload to youtube for us to debug, I think it is the easiest way. Or paste more figures.

Best

hanabhp, 2020-10-16:
Thank you very much @yujianyuanhaha , I will get back to you soon!

hanabhp, 2020-11-09:
Hi @yujianyuanhaha 

I can't set the channel manually using `nexutil -k` since it throws the driver error with invalid channel though the interface is up, and obviously that's why no packet is captured using `tcpdump`.
Please see the slideshow below with all the steps and the current set-up that I have. 

Thank you very much!

https://youtu.be/Q88Wwd-Ebnk

yujianyuanhaha, 2020-11-13:
@hanabhp i see, on the transmitter router side(I assume you have a ASUS router as transmitter and another router as receiver), simply manually set the channel and bandwidth should work.

hanabhp, 2020-11-15:
thank you I appreciated your reply @yujianyuanhaha. 
my question is what do you refer to as 'manually setting' the bandwidth and the channel on the transmitter router?

I set the channel and bandwidth to the same values on both routers using GUI (w/ nexmon (i.e., receiver, the one that is supposed to sniff the packets received on the set channel) and w/o nexmon (i.e., transmitter, the one that receives iperf packets from the laptop and broadcast them on the set channel)).

When I check the channel value on both routers using ```wl channel``` it outputs the same value as I the one I have set using router's GUI.
What am I missing?!

yujianyuanhaha, 2020-11-16:
@hanabhp
manually setting mean use web GUI (default is 192.168.50.1 in browser), see attached below.

[image]

Best.

hanabhp, 2020-11-16:
Thank you @yujianyuanhaha, I have done this. I later figured out the problem was with the transmitting device (Linux machine) which didn't transmit on the channel and the bandwidth that I've set for both the routers!

closed 2020-11-16 by hanabhp