seemoo-lab/nexmon_csi #133: socket error (93: Protocol not supported)

yq11, 2020-10-16:


yq11, 2020-10-16:
Greetings!
When I am using nexutil, how to solve the following error：
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: bad file descriptor)
ERR: no vaild answer receview

zeroby0, 2020-10-16:
Did `ifconfig wlan0 up` work? Also, which commit of nexmon_csi are you using?

yq11, 2020-10-16:
> 做了`ifconfig wlan0 up`工作？另外，您使用的是哪个nexmon_csi提交，您的内核版本是什么？

Thanks for your reply.This problem has been resolved, the patch needs to be reinstalled after each restart.
Now when I use iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
ifconfig mon0 up,
that is : command failed: Operation not supported(-95)
   Looking forward to your reply

zeroby0, 2020-10-16:
`ifconfig mon0 up` fails if it's already up. You can ignore the error if that's the case.

yq11, 2020-10-16:
> `ifconfig mon0 up` fails if it's already up. You can ignore the error if that's the case.

Thank you very much . I'm going to try it from the beginning again. Thank you for your answer

yq11, 2020-10-17:
> `ifconfig mon0 up` fails if it's already up. You can ignore the error if that's the case.

Hello, first of all, your work is undoubtedly very successful, I have successfully caught the packet.

But why can't I continue to get packets after receiving a certain size? And the packet cannot be retrieved after the second restart (the router is sending it all the time)

zeroby0, 2020-10-17:
There is a bug somewhere that causes the firmware to crash after running for sometime. You need to reload the driver after restarting. This can be automated, though.

Fixes for both of them are available on my fork: https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97.

yq11, 2020-10-20:
> 某个地方存在一个错误，导致一段时间运行后固件崩溃。重新启动后，需要重新加载驱动程序。不过，这可以自动化。
> 
> 在我的fork上都可以找到针对它们的修复程序：[https](https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97) : [//github.com/zeroby0/nexmon_csi/tree/pi-4.19.97](https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97)。

Sorry to bother you again. When I re-operate as above,use :nexutil -Iwlan0 -s500 -b -l34 -vm, The result is:
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

when I reboot, use : ifconfig wlan0 up, then : 
wlan0: ERROR while getting interface flags: No such device

zeroby0, 2020-10-21:
> when I reboot, use : ifconfig wlan0 up, then :
wlan0: ERROR while getting interface flags: No such device

This happened to me when the nexmon firmware I installed is corrupt.

> When I re-operate as above,use :nexutil -Iwlan0 -s500 -b -l34 -vm, The result is:
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

Which version of Nexmon_CSI are you using?

yq11, 2020-10-21:
> This happened to me when the nexmon firmware I installed is corrupt.
>
>
> Which version of Nexmon_CSI are you using?

use it: wget https://raw.githubusercontent.com/zeroby0/nexmon_csi/pi-4.19.97/install.sh -O install.sh

zeroby0, 2020-10-21:
I'll install and check.

yq11, 2020-10-21:
> I'll install and check.

Thanks for your help very much.  I guess because I am in China, the script cannot access some websites to download some data packages.  If your check is ok, it should be my network problem

zeroby0, 2020-10-23:
So I installed it and it ran ok. I don't know why it failed for you.

Alternatively, the 5.4 kernel version of nexmon_csi is now available. 

https://github.com/zeroby0/nexmon_csi/tree/pi-5.4.51

This for has amsdu and null check fixes out of the box which cause the original version to crash.

yq11, 2020-10-23:
> So I installed it and it ran ok. I don't know why it failed for you.
> 
> Alternatively, the 5.4 kernel version of nexmon_csi is now available.
> 
> https://github.com/zeroby0/nexmon_csi/tree/pi-5.4.51
> 
> This fork has AMSDU frame and null check fixes out of the box which cause the original version to crash.

OK，I will try again. Maybe I have a problem somewhere.  Thanks for you again.  I will use this great project in later experiments and papers.

pk44-p, 2021-02-09:
Hi, I have the same issue:

I have used this: wget https://raw.githubusercontent.com/zeroby0/nexmon_csi/pi-5.4.51/install.sh -O install.sh

when I use nexutil, I get the following error:

nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

How can I solve this? I have tried re-installing everything but it does not work.

Thanks

zeroby0, 2021-02-13:
@pk44-p 

As you suggested in the email, it is because the kernel is upgraded to 5.10 when running apt upgrade. 

I installed raspios_lite_armhf-2021-01-12 ([permalink](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-01-12/)). And then used this: https://github.com/nexmonster/nexmon_csi/tree/pi-4.19.97-plus, except I used the 5.4.51 script - `wget https://raw.githubusercontent.com/zeroby0/nexmon_csi/pi-5.4.51/install.sh -O install.sh` instead of the 4.19.97 installation script. I just installed it on my Pi 4, and can confirm that it still works. :)

juliocza7, 2021-03-04:
I have same issue:

nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

I install and reinstall wget https://raw.githubusercontent.com/zeroby0/nexmon_csi/pi-5.4.51/install.sh -O install.sh on raspios_lite_armhf-2021-08-20 I used kernel 5.4 but have same issue. any solution?

pk44-p, 2021-03-04:
Hi @juliocza7, I solved the issue, I followed the suggestion from @zeroby0 comment.
1) I re-intalled the 2021-01-11-raspios-buster-armhf-full
2) then used this: https://github.com/nexmonster/nexmon_csi/tree/pi-4.19.97-plus, except I used the 5.4.51 script - wget https://raw.githubusercontent.com/zeroby0/nexmon_csi/pi-5.4.51/install.sh -O install.sh instead of the 4.19.97 installation script. 

Hope it works for you!

zeroby0, 2021-03-04:
The kernel upgraded to 5.10, and the Nexmon_csi is not compatible with the new version. I'm supposed to update the Readme on the repo, but I forgot. You can follow the instructions on the 5.4.51 branch, and get headers as described on the 4.19.97 branch, **without** running apt upgrade. That will keep you on the kernel with 2021-01-11 and get you it's headers from source, and Nexmon_csi will work. 

I will update the Readme tonight or tomorrow.

juliocza7, 2021-03-10:
Thank you @pk44-p and @zeroby0 I solved the issue, but only work 5.4.51 version. I tried install 5.4.51-plus version from zeroby0 but don’t work for me. Now, I use python explorer but those scripts only work when capture more bytes like rssi (5.4.51-plus version), any suggestions how get rssi or modify python scripts for 5.4.51 version?

zeroby0, 2021-03-10:
https://www.github.com/nexmonster/nexmon_csi/tree/feature%2Fpython/utils%2Fpython

I made a csi-explorer for the 5.4.51 branch.

juliocza7, 2021-03-13:
@zeroby0 thank!!! I use your csi-explorer python and work.

I use two Raspberry Pi, one pi3b+ makes ping to my PC and I setup this pi3B+ like a router with channel 157 - 20MHz bandwidth, is necessary setup bandwidth to 80MHz or can I use 20MHz? 
When I captured CSI-packet with other pi4B used mcp -c 157/80 -C 1 -N 1 but my bandwidth is 20MHz and I get packets on channel 157 with 20MHz bandwidth but I don't know if this packets are useful for my analysis because use 20MHz and not 80MHz.
Thank you!!!