seemoo-lab/nexmon_csi #134: read CSI value from bytes on ASUS router

yujianyuanhaha, 2020-10-19:
greetings,

Recently I plan to write a bit my own code to **convert these bytes (using Wireshark tools) to CSI value**. It is clear to me how to convert the data into Csi value on NEXUS phone by description [analyzing-the-csi](https://github.com/seemoo-lab/nexmon_csi#analyzing-the-csi). 

However, it is not clear to me on ASUS router. The description said " _The bcm4358 and bcm4366c0 return values in a floating point format with one bit sign of the following nine or twelve bits of a real part and the same for an imaginary part, followed by an exponent of five or six bits._" 

My understanding is it should be "_The bcm4358 and bcm4366c0 return values in a floating point format with one bit sign of the following **nine** or **ten** bits of a real part and the same for an imaginary part, followed by an exponent of **six** or **five** bits._" in this way number of bit is exact 32 per complex value.

Let me know if  I get it wrong @DanielAW @schmittner @zeros @mzakharo @zeroby0

Best

zeroby0, 2020-10-19:
The example matlab code in utils folder might help.

yujianyuanhaha, 2020-10-19:
@zeroby0 thanks. I have to read csi from byte cuz I have to deal with new format `.pcap` using the path in [CSI with RSSI on ASUS router](https://github.com/seemoo-lab/nexmon_csi/issues/126), which nexmon matlab code does NOT work

Deltaaaaaaaaaa, 2020-10-25:
@yujianyuanhaha 
Hi !
By observing the HEX code, I found the complex float is packed as follow.
2bits 0, 1bit real sign, 11bits real, 1bit img sign, 11bits img, 6bits exp
Where exp is a integer consisted of 1bit sign and 5bits digits, the real and imag parts should be multiplied by (2^exp).
However, unless you are required to process the data in other languages( e.g. python), I don't recommand you to write your own unpack code, since the official matlab code is highly effiecient (much more efficient compared to the python code I wrote). 
Modifying the readpacp.m file (somewhere around line 57) would be a better way to get your work done.
Good luck!

yujianyuanhaha, 2020-12-20:
@Deltaaaaaaaaaa  thanks a lot, however, the RSSI I get is in 2 bytes only, detail explain [here](https://github.com/seemoo-lab/nexmon_csi/issues/126#issuecomment-705074704) ....