seemoo-lab/nexmon_csi #137: On Rasp pi 4B, tcpdump get no capture but 1 received

takaya0111, 2020-10-29:
Dear all, I use this tool on Raspi 4. When I use
`tcpdump -i wlan0 dst port 5500`
output: 
0 packets captured
1 packet received by filter
0 packets dropped by kernel
or 
0 packets captured
0 packet received by filter
0 packets dropped by kernel

Is this correctly worked? I stored by
 `tcpdump -i wlan0 dst port 5500 -c 10 -w  "filename.pcap" -vv` 
and inspect by tshark , the pcap file seemed to be empty.

Many thanks.

zeroby0, 2020-10-29:
The command is correct. Assuming you have run set the makecsiparams 's parameters correctly, and have made sure that your router is operating in the same channel you're listening to, maybe the firmware is crashing.

Can you post the contents of `cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics`  just after starting csi collection?

takaya0111, 2020-10-30:
Thank you!  I used `cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics` after `tcpdump -i wlan0 dst port 5500`,
output was
` BCM4345 r6 @ 37.4/160.2/160.2MHz
000000.001 nvram_init: called again without calling nvram_exit()
000000.002 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.002 nvram_init: called again without calling nvram_exit()
000000.147 reclaim section 0: Returned 41892 bytes to the heap
000000.149 wlc_bmac_info_init: host_enab 0
000000.149 Overriding vendor id = 0x14e4
000000.149 Overriding device id = 0x43ab
000000.149 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.199 wl0: wlc_stf_txcore_shmem_write: No clock
000000.200 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.205 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.206 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.206 TCAM: 256 used: 237 exceed:0
000000.207 reclaim section 1: Returned 118076 bytes to the heap
000000.207 reclaim section 4: Returned 44 bytes to the heap
000000.208 sdpcmd_dpc: Enable
000000.230 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.232 wl0: unable to find iovar "rsdb_mode"
000000.232 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.232 wl0: unable to find iovar "tdls_enable"
000000.232 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000000.921 wl0: unable to find iovar "toe_ol"
000000.921 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000000.921 wl0: wl_open
000000.932 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000000.952 wl0: unable to find iovar "nd_hostip_clear"
000000.952 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
000315.140 wl0: unable to find iovar "toe_ol"
000315.140 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000315.152 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex`

CristianTuretta, 2020-10-30:
Dear developers,
I have a similar issue on my Raspberry 4 Model B 4GB RAM, after configuration I followed the _usage_ section in your guide.
When I prompt ` tcpdump -v -i wlan0 dst port 5500` the output is `tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes`. After waiting, I stopped the capture (`ctrl C`) and the output was the following:

`0 packets captured`
`0 packets received by filter`
`0 packets dropped by kernel`

Then, reading this issue I tried `cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics` and the output was:

000000.345 wl0: unable to find iovar "toe_ol"
000000.345 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000000.345 wl0: wl_open
000000.356 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000000.380 wl0: unable to find iovar "nd_hostip_clear"
000000.381 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
000011.848 wl0: link local addresses being set! watch out!!
001223.451 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
001243.826 wl0: unable to find iovar "toe_ol"
001243.826 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
001611.891 wl0: dma_rx: bad frame length (1862)
001614.435 wl0: dma_rx: bad frame length (1666)
001787.567 wl0: dma_rx: bad frame length (1790)
001917.226 wl0: dma_rx: bad frame length (1650)
002150.457 wl0: dma_rx: bad frame length (1898)
002202.033 wl0: dma_rx: bad frame length (1754)
002478.054 wl0: dma_rx: bad frame length (1666)
002774.833 wl0: dma_rx: bad frame length (1694)
003025.783 wl0: dma_rx: bad frame length (1850)
003449.541 wl0: dma_rx: bad frame length (1898)
003868.512 wl0: dma_rx: bad frame length (1714)
003932.183 wl0: unable to find iovar "nd_hostip_clear"
003932.183 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
003932.258 wl0: wl_open
003932.270 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
003949.959 wl0: wlc_iovar_op: pfn_gscan_cfg BCME -23 (Unsupported)
003957.500 wl0: wlc_iovar_op: pfn_gscan_cfg BCME -23 (Unsupported)
003965.035 wl0: wlc_iovar_op: pfn_gscan_cfg BCME -23 (Unsupported)
003969.236 wl0: unable to find iovar "nd_hostip_clear"
003969.236 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
003979.396 wl0: link local addresses being set! watch out!!
004407.289 dngl_finddev: vslave 0 not found
004407.290 dngl_finddev: vslave 0 not found
004407.293 dngl_finddev: vslave 0 not found
004420.935 wl0: unable to find iovar "toe_ol"
004420.935 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
004460.276 wl0: dma_rx: bad frame length (1658)

What should I do? I'm connected to raspberry via SSH and my raspberry is connected with the router via ethernet. Thanks

zeroby0, 2020-10-30:
@takaya0111 @CristianTuretta 

Did you restart the Pi after installing nexmon_csi? If you did, the modified firmware needs to be reloaded again on reboot.

Try starting over again but following instructions from one of these branches:
|Kernel version| link|
|---|---|
|5.4.51| https://github.com/zeroby0/nexmon_csi/tree/pi-5.4.51 |
|4.19.97| https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97|

The 5.4.51 version is what I'm using right now, and the 4.19.97 version is what I was using the last 6 months. Both of them setup firmware persistence over reboots, and have been tested in various settings.

CristianTuretta, 2020-10-30:
@zeroby0 thanks for your reply. I had followed the guide for kernel version `5.4.51`. After Installing I have rebooted the raspberry. Then I followed the _usage_ guide, but still 0 captured/received/dropped. I tried to run `cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics`  and got the following:

 BCM4345 r6 @ 37.4/160.2/160.2MHz
000000.001 nvram_init: called again without calling nvram_exit()
000000.002 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.002 nvram_init: called again without calling nvram_exit()
000000.147 reclaim section 0: Returned 41892 bytes to the heap
000000.149 wlc_bmac_info_init: host_enab 0
000000.149 Overriding vendor id = 0x14e4
000000.149 Overriding device id = 0x43ab
000000.150 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.173 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.200 wl0: wlc_stf_txcore_shmem_write: No clock
000000.201 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.206 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.207 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.207 TCAM: 256 used: 237 exceed:0
000000.208 reclaim section 1: Returned 118076 bytes to the heap
000000.208 reclaim section 4: Returned 44 bytes to the heap
000000.209 sdpcmd_dpc: Enable
000000.237 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.239 wl0: unable to find iovar "rsdb_mode"
000000.239 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.241 wl0: unable to find iovar "tdls_enable"
000000.241 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000072.911 wl0: unable to find iovar "toe_ol"
000072.911 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000072.912 wl0: wl_open
000072.923 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000072.942 wl0: unable to find iovar "nd_hostip_clear"
000072.942 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
000085.859 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000100.241 wl0: unable to find iovar "toe_ol"
000100.241 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)

takaya0111, 2020-10-30:
@zeroby0  I checked Kernel version and version was 5.4.72, not 5.4.51.  Does the guide is appiled for  version 5.4.72 ?

zeroby0, 2020-10-30:
@takaya0111 I see. Great find! Then maybe the kernel upgraded and broke something. If you don't run apt upgrade, then the version will stay the same. You will have to get kernel headers in the way shown in 4.19.97 repo.

takaya0111, 2020-10-30:
@zeroby0 Thanks. I followed the guide for 4.19.97. After "getting started " and do `ifconfig wlan0 up`,
wlan0: ERROR while getting interface flags: No such device
What should I do?

In addition ,while doing `rpi-source`, questions below appeared and I all answered y. Is it correct?

Code coverage for fuzzing (KCOV) [N/y/?] (NEW) 
  Enable comparison operands collection by KCOV (KCOV_ENABLE_COMPARISONS) [N/y/?] (NEW) 
  Instrument all code by default (KCOV_INSTRUMENT_ALL) [Y/n/?] (NEW)

zeroby0, 2020-10-31:
> `Code coverage for fuzzing (KCOV) [N/y/?] (NEW)`

I just go with the default value (The capitalized letter in [N/y/?]). 

> `wlan0: ERROR while getting interface flags: No such device`

This feels a lot like firmware is not loaded by the kernel. Can you run lsmod and see if brcmfmac is in the list?
You're probably installing 4.19.97 on a later image of buster which has a 5.x kernel. The instructions for 4.19.97 has a link to download the correct version.

I just installed 4.19.97 and it works just fine, even across reboots. Do check and let me know if that helps.

takaya0111, 2020-10-31:
I reinstalled 4.19.97 and answered N in [N/y/?], error about 
`wlan0: ERROR while getting interface flags: No such device`
 resolved. But when `tcpdump -i wlan0 dst port 5500`, get 0 paclets.

By `cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics` after that, output:

000.001 
RTE (SDIO-CDC) 7.45.189 (r714228 CY) on BCM4345 r6 @ 37.4/160.2/160.2MHz
000000.001 nvram_init: called again without calling nvram_exit()
000000.002 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.002 nvram_init: called again without calling nvram_exit()
000000.146 reclaim section 0: Returned 41892 bytes to the heap
000000.148 wlc_bmac_info_init: host_enab 0
000000.148 Overriding vendor id = 0x14e4
000000.148 Overriding device id = 0x43ab
000000.149 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.171 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.198 wl0: wlc_stf_txcore_shmem_write: No clock
000000.199 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.204 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.205 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.206 TCAM: 256 used: 237 exceed:0
000000.207 reclaim section 1: Returned 118076 bytes to the heap
000000.207 reclaim section 4: Returned 44 bytes to the heap
000000.207 sdpcmd_dpc: Enable
000000.229 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.231 wl0: unable to find iovar "rsdb_mode"
000000.231 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.232 wl0: unable to find iovar "tdls_enable"
000000.232 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000100.074 wl0: unable to find iovar "toe_ol"
000100.074 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000100.074 wl0: wl_open
000100.086 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000100.101 wl0: unable to find iovar "nd_hostip_clear"
000100.101 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
000118.287 wl0: unable to find iovar "toe_ol"
000118.287 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)

and by `ifconfig`, output about wlan0 is
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 "xxxxxxxxxxxxxxx"  prefixlen 64  scopeid 0x20<link>
        ether "xxxxxxxxxxxxxx"  txqueuelen 1000  (Ethernet)
        RX packets 6203  bytes 1973543 (1.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 2882 (2.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

zeroby0, 2020-10-31:
@takaya0111 Can you post your `uname -a`. And also the output of `dmesg`.

takaya0111, 2020-10-31:
Output of `uname -a` 
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

Output of `dmesg`
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.97-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1294 SMP Thu Jan 30 13:21:14 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 1864704 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
[    0.000000] pcpu-alloc: s36928 r8192 d24512 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2059584
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:BF:6B:2E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 7893004K/8245248K available (8192K kernel code, 687K rwdata, 2408K rodata, 2048K init, 850K bss, 90100K reserved, 262144K cma-reserved, 7458816K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 688 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28692 entries in 85 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000022] Switching to timer-based delay loop, resolution 18ns
[    0.000257] Console: colour dummy device 80x30
[    0.000726] console [tty1] enabled
[    0.000786] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000822] pid_max: default: 32768 minimum: 301
[    0.001113] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001147] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001967] CPU: Testing write buffer coherency: ok
[    0.002392] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003050] Setting up static identity map for 0x200000 - 0x20003c
[    0.003228] rcu: Hierarchical SRCU implementation.
[    0.004105] smp: Bringing up secondary CPUs ...
[    0.004978] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005938] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006865] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007001] smp: Brought up 1 node, 4 CPUs
[    0.007069] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.007094] CPU: All CPU(s) started in HYP mode.
[    0.007115] CPU: Virtualization extensions available.
[    0.007967] devtmpfs: initialized
[    0.018404] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.018638] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.018680] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.026236] pinctrl core: initialized pinctrl subsystem
[    0.027177] NET: Registered protocol family 16
[... 298 lines cut ...]
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
snd_bcm2835            24576  1
snd_pcm               102400  4 vc4,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core
bcm2835_codec          36864  0
snd_timer              32768  1 snd_pcm
bcm2835_v4l2           45056  0
snd                    73728  7 snd_compress,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
v4l2_mem2mem           24576  1 bcm2835_codec
[... cut ...]
videobuf2_common       45056  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              200704  6 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
media                  36864  3 bcm2835_codec,videodev,v4l2_mem2mem
vc_sm_cma              36864  1 bcm2835_mmal_vchiq
rpivid_mem             16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  454656  24

zeroby0, 2020-10-31:
I missed the resolved part in `error about
wlan0: ERROR while getting interface flags: No such device
resolved.`

I think it should work now. Make sure you have set the correct channel, and are running the usage commands as root. Can you run these and post the output?
`sudo ifconfig wlan0 up`
`sudo iwlist wlan0 scan`

This will print out the wifi networks. You may remove the Mac IDs of your router if you prefer to.

takaya0111, 2020-10-31:
The output was 

`wlan0　　 No　scan results.`

Do I have to delete Mac IDs?

zeroby0, 2020-10-31:
It happens sometimes. Reboot and run 
`sudo ifconfig wlan0 up`
`sudo iwlist wlan0 scan`

before running anything else. Starting csi collection disables scanning.

takaya0111, 2020-10-31:
Thanks. After reboot, output was:

wlan0     Scan completed :
          Cell 01 - Address: 50:C4:DD:47:6A:27
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-28 dBm  
                    Encryption key:on
                    ESSID:"Buffalo-G-6A26"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 9 Mb/s
                              18 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 12 Mb/s; 24 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 100ms ago
                    IE: Unknown: 000E42756666616C6F2D472D36413236
                    IE: Unknown: 010882848B961224486C
                    IE: Unknown: 030101
                    IE: Unknown: 2A0104
                    IE: Unknown: 32040C183060
                    IE: Unknown: 2D1AAC0117FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601000500000000000000000000000000000000000000
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 7F080000000000000000
                    IE: Unknown: 0B0503002A127A
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: 07064A5020010D10
                    IE: Unknown: DD830050F204104A0001101044000102103B000103104700102880288028801880A88050C4DD476A271021000C42756666616C6F20496E632E1023000A5743522D313136364453102400065254323836301042000831323334353637381054000800060050F20400011011000357505310080002268C103C0001011049000600372A000120
                    IE: Unknown: DD07000C4303000000
                    IE: Unknown: DD09000740010109000000
          Cell 02 - Address: 1C:3B:F3:F2:6D:53
                    Channel:5
                    Frequency:2.432 GHz (Channel 5)
                    Quality=70/70  Signal level=-34 dBm  
                    Encryption key:on
                    ESSID:"TP-Link_6D53"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 100ms ago
                    IE: Unknown: 000C54502D4C696E6B5F36443533
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030105
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AEE111BFFFFFF0000000000000000000100000000000000000000
                    IE: Unknown: 3D16050C0600000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080100000000000040
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD800050F204104A0001101044000102103B00010310470010876543219ABCDEF012341C3BF3F26D531021000754502D4C696E6B1023000941726368657220433610240003312E301042000C4172636865722043362076321054000800060050F20400011011000A417263686572433676321008000200041049000600372A000120
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
          Cell 03 - Address: 98:F1:99:B9:31:5E
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=64/70  Signal level=-46 dBm  
                    Encryption key:on
[... 353 lines cut ...]
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 100ms ago
                    IE: Unknown: 000E42756666616C6F2D412D36413236
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 030140
                    IE: Unknown: 07064A5020241314
                    IE: Unknown: 050400010000
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD310050F204104A0001101044000102104700102880288028801880A88050C4DD476A27103C0001021049000600372A000120
                    IE: Unknown: 2D1A6F0017FFFF0000010000000000000000000000000C1406A60800
                    IE: Unknown: 3D1640070400000000000000000000000000000000000000
                    IE: Unknown: BF0C3028C131FAFF0C03FAFF0C03
                    IE: Unknown: C005013A00FAFF
                    IE: Unknown: 7F080000000000000040
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: 200100
                    IE: Unknown: C304020F0F00
                    IE: Unknown: DD07000C4304000000
                    IE: Unknown: DD07000CE700000000
                    IE: Unknown: DD09000740010109000000

zeroby0, 2020-10-31:
I'm assuming `ESSID:"Buffalo-A-6A26"` is your router.

So you're running
```
sudo su
mcp -c 64/80 -C 1 -N 1
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -vOuMBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
iw dev wlan0 interface add mon0 type monitor
ip link set mon0 up

tcpdump -i wlan0 dst port 5500
```

right?

Also, I'm seeing the Pi being unable to collect CSI in higher channels in some cases. Can you change settings in your router to channel 36, bandwidth 80 and try?

takaya0111, 2020-10-31:
I don't know much about wi-fi,　does "in higher channels"  indicate wi-fi in the 5ghz band? Does that mean that a wi-fi in the 2.4Ghz band will work properly then?

zeroby0, 2020-10-31:
Yes, I'm referring to 5GHz band. Some channels in 2 GHz band don't work too (https://github.com/seemoo-lab/nexmon_csi/issues/138#issuecomment-719947619), so let's stick to 5 GHz.

You log into your router, and go to WiFi advanced settings, and change channel to 36. You can find a list of channels here: https://en.wikipedia.org/wiki/List_of_WLAN_channels

In my case, my router was originally set to channel 149, and I was not able to collect any csi data. But then I changed it to channel 36, and now I can collect data.

takaya0111, 2020-10-31:
I changed channel to 36 and after reset router and reboot, run 
`sudo su
mcp -c 36/80 -C 1 -N 1
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -vOuMBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
iw dev wlan0 interface add mon0 type monitor
ip link set mon0 up

tcpdump -i wlan0 dst port 5500`

 but , captured 0 packets. 
Should I change to 2.4Ghz wi-fi?

zeroby0, 2020-10-31:
`nexutil -Iwlan0 -s500 -b -l34 -vOuMBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==` is for channel 64. You should use the output generated by mcp.

takaya0111, 2020-10-31:
I could receive the packet! Thanks for staying with me for so long.

closed 2020-10-31 by takaya0111

zeroby0, 2020-10-31:
Nice!

So I figured out why your previous channel didn't work. For some reason, some channel/bandwidth combinations are not allowed. You can override this setting by modifying https://github.com/zeroby0/nexmon_csi/blob/pi-4.19.97-plus/src/regulations.c and adding your own chanspec.

Below line 217, there are chanspecs for 36/80 and 157/80. I changed my router to 157/80 and it worked nicely again. I don't know how to generate chanspec for arbitrary combinations, but I'm trying to find out.