seemoo-lab/nexmon_csi #138: Did CSI contain Network card MAC or Router MAC  ??

John-tw, 2020-10-31:


zeroby0, 2020-10-31:
The MAC address of wherever the packet is coming from. So, both.

It's just copied from Address 2 field of 802.11 frame.
https://www.oreilly.com/library/view/80211-wireless-networks/0596100523/ch04.html

John-tw, 2020-10-31:
@zeroby0  thx your commented ,
I want get this AP MAC(Router MAC),
[image]
but only 3 packages contain this AP MAC(Router MAC) in my 500 packages.
most other is my Network card MAC
Did you know that how to get more AP MAC(Router MAC)??

zeroby0, 2020-10-31:
Play a youtube video on your laptop, and set makecsiparams `-b 0x88 -m 68:ff:7b:b6:25:14`. That way, the data packets from router to laptop will be captured, and other communication will not be. 

0x88 (0b10001000) is QoS Data frames. You can find a list of frame types here: https://en.wikipedia.org/wiki/802.11_Frame_Types

John-tw, 2020-10-31:
@zeroby0 can I use ping <my router up>?

zeroby0, 2020-10-31:
Sure.

John-tw, 2020-10-31:
@zeroby0   hi, 
I find my problem.
My router is use 2.4GHz . 
But my params  ""./makecsiparams -c 157/80 -C 1 -N 1""  is wrong.  I am stupefy,hahaha
I will find correct params with 2.4GHz,  Maybe
thx bro

zeroby0, 2020-10-31:
Happened to me too haha.
You can use `sudo ifconfig wlan0 up` and `sudo iwlist wlan0 scan` to scan wifi networks. Then use `mcp -c channel/20 -C 1 -N 11

John-tw, 2020-10-31:
@zeroby0    okay ,  channel is 6
[image]

zeroby0, 2020-10-31:
https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/src/regulations.c#L195-L219

I think that's because 6/20 is not listed here. Try adding it.

John-tw, 2020-11-01:
@zeroby0 
Sor bor !
I don't know that how to  translate "6/20" .   Like as 0xe02a，// 36/80.
what's language???
[image]

zeroby0, 2020-11-01:
I don't know either. But you can use the other thing: `CH20MHZ_CHSPEC(6)`, above that line.

John-tw, 2020-11-01:
@zeroby0 , bor
https://github.com/seemoo-lab/nexmon_csi/issues/2#issuecomment-539497815
I use "nexutil -k6/20 && nexutil -k "  , get  "0x1006"
[image]
and this is my code ( after change)
[image]
It still not work for collect 2.4GHz CSI

zeroby0, 2020-11-01:
Don't know why. Instead of adding new channels to nexmon_csi, may be try changing your router to an already present channel. Maybe 7?

John-tw, 2020-11-01:
@zeroby0   maybe 
i will try

John-tw, 2020-11-02:
@zeroby0  hi bor,
Tell you a good news!!
I get 2.4GHz CSI data  ( 2.4Gh and channel 7)
This picture is my architecture diagram.
[image]
I use pi4
my csi version is https://github.com/zeroby0/nexmon_csi/tree/pi-5.4.51

zeroby0, 2020-11-02:
Nice!

yjxb1, 2023-09-19:
> Play a youtube video on your laptop, and set makecsiparams `-b 0x88 -m 68:ff:7b:b6:25:14`. That way, the data packets from router to laptop will be captured, and other communication will not be. [...]

Hello, I checked the information on this website, but I am very confused about one thing. As you said 0X88 (0b10001000), how does it correspond to this frame structure? Does 1 in this byte represent frame structure B7 or B0?