seemoo-lab/nexmon_csi #139: Can receive any packet using tcpdump

Colin-Guolin, 2020-10-31:
[image]

I am new to Linux,  I installed the nexmon on my raspberry. when I try to run the commend "_tcpdump -i wlan0 dst port 5500_", it cannot capture any packets, I attached a screen shot here, could anyone help me with this? I followed instruction and killed WiFi connection, how would it know which router it should connect to and which router's CSI is the one that I want it to capture?

zeroby0, 2020-11-01:
Hey Colin!

> I followed instruction and killed WiFi connection, how would it know which router it should connect to and which router's CSI is the one that I want it to capture?

The Raspberry Pi will not connect to any router. Instead, it sniffs all traffic on the wireless channel you configured makecsiparams with. Here is a list of WiFi channels: https://en.wikipedia.org/wiki/List_of_WLAN_channels

You can set your WiFi channel in your router's advanced settings. I recommend you set it to channel 36, as other channels seem to have a problem with CSI collection. Then you can run the following commands, and your Pi will collect CSI for all packets in channel 36.

```
sudo su
makecsiparams -c 36/80 -C 1 -N 1
pkill wpa_supplicant
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -vKuABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
ifconfig mon0 up

tcpdump -i wlan0 dst port 5500
```

Also please go through previous issues: https://github.com/seemoo-lab/nexmon_csi/issues?q= They have a lot of information on nexmon_csi and how to debug issues.

Colin-Guolin, 2020-11-01:
> Hey Colin!
>
>
> The Raspberry Pi will not connect to any router. Instead, it sniffs all traffic on the wireless channel you configured makecsiparams with. Here is a list of WiFi channels: [...]

Thanks a lot, I got it run. But I have another problem now.

I can use those commend to capture the packet, but when I stop it by using "Ctrl + C", and run the commend "tcpdump -i wlan0 dst port 5500" again. My pi cannot capture any packet again, and run "iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor" it shows me "commend failed: Operation not supported (-95)".
reboot and run those commend again can solve the problem. I wondering why and how to solve this problem?

thanks a lot

Colin-Guolin, 2020-11-02:
> Hey Colin!
>
>
> The Raspberry Pi will not connect to any router. Instead, it sniffs all traffic on the wireless channel you configured makecsiparams with. Here is a list of WiFi channels: [...]

Thanks a lot, problem solved