seemoo-lab/nexmon_csi #142: Stuck on the step 6

wangtengyi5555, 2020-11-09:
I always met the problem of the platform not supported. I thought it was because of the hostname, however, even if I changed the hostname or the setup_env.sh, it showed me "platform not supported" or "source: command not found", based on the different commands or different changes I made to the **setup_env.sh**. I tried sudoing command, #!/bin/bash and #!/bin/sh to the setup_env.sh, tried to use (sudo) **.** or source the shell, none of them worked.

My operation system is Linux ubuntu 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:06:14 UTC 2016 i686 i686 i686 GNU/Linux (Ubuntu 16.04 LTS) downloaded from http://old-releases.ubuntu.com/releases/16.04.4/, while another XUbuntu with Linux ubuntu 4.15.0-45-generic #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:19 UTC 2019 i686 i686 i686 GNU/Linux downloaded from http://cdimage.ubuntu.com/xubuntu/releases/16.04/release/. Both could not work. I downloaded both of them of the first image.

error messages types:
1. Platform not supported!
2. sudo: source: command not found
3. 
[image]
if I add #!/bin/sh to the setup_env.sh, and use .

zeroby0, 2020-11-09:
You should run Nexmon on a Raspberry Pi.
And it may not work on some kernel versions even on Raspberry OS. The setup which worked for me is here: https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97

hanabhp, 2020-11-11:
I faced with the same error when installing nexmon on the ASUS router. In my case it was an easy fix, I just had to install `g++` and it worked.

wangtengyi5555, 2020-11-15:
Thank you very much 😊