seemoo-lab/nexmon_csi #144: The CSI value of the second antenna is 0

yangtat, 2020-11-16:
I tried to use one of the Nexus 6P antennas to collect the CSI of the router's two antennas. But  CSI with one antenna is always 0.
I used:  `makecsiparams -c 36/20 -C 1 -N 3 -m  mac`.
Is it my parameter problem?

yujianyuanhaha, 2020-11-16:
@yangtat 
I also met the problem with a pair of ASUS router, each with 4 antennas, while one antenna mostly is 0.

Try to move the mobile phone around, it may have sth to do with the position, somewhere else may have better coverage.

yangtat, 2020-11-17:
@yujianyuanhaha 
I tried to move the location of the phone, but it didn't work.
I'm trying to get CSI from one antenna,
`makecsiparams -c 36/20 -C 1 -N  1 -m`
`makecsiparams -c 36/20 -C 1 -N  2 -m`
`makecsiparams -c 36/20 -C 1 -N  4 -m`
`makecsiparams -c 36/20 -C 1 -N  8 -m`
But only in the first case can I collect CSI.
Is there a problem with my parameters?Or something else?
When I was collecting CSI, I used my laptop to ping the router.The router has two 2.4g antennas and two 5G antennas.

zeroby0, 2020-11-17:
Can you try with:
`makecsiparams -c 36/20 -C 1 -N 3 -m`

Usually the 2.4GHz and 5GHz antennas are placed alternatively to reduce interference. So maybe antennas 1 and 3 are 5 GHz and antennas 2 and 4 are 2.4 GHz.

yangtat, 2020-11-17:
@zeroby0  
I tested each antenna individually, but only the first one collected data.
Does my router only have one antenna working?

zeroby0, 2020-11-17:
I don't know yet.

The command you posted has 4 and 8, but not 3, so I wanted to make sure you tried 3 as well.

yangtat, 2020-11-18:
@zeroby0 
I used Linux 802.11n CSI Tool to collect the CSI from the two antennas of the router successfully. 
There seems to be nothing wrong with my router.
As I understand it, 3=b0011, and `-N 3` means to collect the first and second spatial stream.
In fact, I have tried many combinations, but I can only collect the first spatial stream.

yangtat, 2020-11-20:
Make sure your router is working under 802.11/n!
My router works in MIX mode, so I can't collect the information of the second antenna

closed 2020-11-20 by yangtat