seemoo-lab/nexmon_csi #145: Neither tcpdump nor wireshark can obtain data.

memory-xm, 2020-11-16:
greetings,
I am using RPI4, bcm434355c0. The whole process is finished, but tcpdump cannot receive data, and it cannot be captured using wireshark. At the same time, after the wireless network card turns on the monitoring mode, the ping command cannot be used to send packets to the wireless gateway. Because my experiment needs to monitor data packets of a specific address in a wireless environment, how can I solve it?

memory-xm, 2020-11-18:
> 问候，
> 我正在使用RPI4，bcm434355c0。整个过程已完成，但是tcpdump无法接收数据，并且无法使用wirehark捕获数据。同时，在无线网卡打开监视模式后，无法使用ping命令将数据包发送到无线网关。由于我的实验需要在无线环境中监视特定地址的数据包，我该如何解决？

tweigel-dev, 2020-12-04:
make sure that you are configure the nexutils right with changing the parameters at the utils/makeparams/makeparams. The channel, bandwith and mac are very important for a successful receive of CSI at nexmon

sangeun-jo, 2023-06-02:
You should install nexmon to monitor data packets of a specific address in a wireless environment.
Then you can't capture CSI. I am tring to find way to do both.