seemoo-lab/nexmon_csi #147: Is there any way to send a data frame from one raspberry pi and make another raspberry pi to receive it

18563407351, 2020-11-19:
Hi guys,
I‘m a beginner to the Linux system. I want to send a data frame from one raspberry pi 4B to another one through WI-FI and measure the CSI using the extractor. It is hard to realize? I'd appreciate it if anyone could give me some advice.

tweigel-dev, 2020-11-22:
you can just ping your device if you are in client mode  or send some dataframes with the monitor mode if your setup requires not connected devices. you only have to install nexmon csi at the receiver to gather csi values.

18563407351, 2020-11-24:
> you can just ping your device if you are in client mode or send some dataframes with the monitor mode if your setup requires not connected devices. you only have to install nexmon csi at the receiver to gather csi values.

Thanks. But how to send data frames between two Raspberry Pis? Does it requires programming in the raspberry pi or just use some commands in the terminal.

zeroby0, 2020-11-24:
The raspberry pi collecting CSI cannot receive any packets.

If you want to collect CSI from the other Raspberry Pi, connect it to your router, and ping your router from the raspberry pi. This will ensure there is WiFi traffic. On your other raspberry pi, start CSI collection with Mac ID filter set to the first raspberry pi.

18563407351, 2020-11-24:
> The raspberry pi collecting CSI cannot receive any packets.
>
> If you want to collect CSI from the other Raspberry Pi, connect it to your router, and ping your router from the raspberry pi. This will [...]

Thanks a lot, I have connected another raspberry pi to the wifi and ping the router. But I still don't know how to use the CSI extractor to collect CSI. I used 'tcpdump  -i wlan0 host 192.138.3.11' to try to capture the packet. No packet is obtained. I'd appreciate it if you can give me some advice.

zeroby0, 2020-11-24:
CSI packets are on UDP Port 5500, so you should collect from that:

```bash
makecsiparams -c channel/bandwidth -C 1 -N 1 -m macIdOfThePiThatPings
# Or optionally, to get CSI of router as well:
# makecsiparams -c channel/bandwidth -C 1 -N 1 -m macIdOfThePiThatPings,macIdOfTheRouter
...
nexutil ...
...
tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000
```

18563407351, 2020-11-24:
> CSI packets are on UDP Port 5500, so you should collect from that:
> 
> ```shell
> makecsiparams -c channel/bandwidth -C 1 -N 1 -m macIdOfThePiThatPings
> # Or optionally, to get CSI of router as well:
> # makecsiparams -c channel/bandwidth -C 1 -N 1 -m macIdOfThePiThatPings,macIdOfTheRouter
> ...
> nexutil ...
> ...
> tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000
> ```

so how can I know the channel the command Ping is using? I use channel 36 and captured the packet as shown below:
[image]

And I think the command ping is using the ICMP protocol, can the CSI extractor collect CSI from the ICMP packet? I'm not sure if the collected CSI is from the ping process.

tweigel-dev, 2020-11-24:
yes it can because ICMP is layer4 and uses layer2(framing).  
Did you gather all ICMP packages or did you also notice a los like mentioned here? @18563407351 
https://github.com/seemoo-lab/nexmon_csi/issues/150

zeroby0, 2020-11-24:
It's very difficult to collect CSI of only ICMP packets. You can see more in #119. The problem is that, the index for the byte which represents ICMP packets keeps changing. And some packets are shorter than that, so accessing that index causes an out of bound error.

Do you want ICMP packets to ensure that all the packets are 80 MHz? You can either send ping packets with a very short interval and trust that most of them are 80 MHz, or you do what I did in https://github.com/zeroby0/nexmon_csi/tree/pi-5.4.51-plus and filter for [frames with type Data](https://en.wikipedia.org/wiki/802.11_Frame_Types). You can use `-b 0xb8` with that modified firmware and it collects all Data Frames from your Raspberry, and most of them are Ping.

18563407351, 2020-11-24:
> yes it can because ICMP is layer4 and uses layer2(framing).
> Did you gather all frames of ICMP packages or did you also notice a leakage like mentioned here? @18563407351
> #150

Thanks. It seems that there is no packet dropped when I used Ping. I haven't met this problem.

18563407351, 2020-11-24:
> It's very difficult to collect CSI of only ICMP packets. You can see more in #119. The problem is that, the index for the byte which represents ICMP packets keeps changing. And some packets are [...]

Thanks very much for your answer. That helps me a lot!

tweigel-dev, 2020-11-24:
Which Setup you are using that you dont have a noticeable loss?

18563407351, 2020-11-24:
> Which Setup you are using that you dont have a noticeable loss?

I'm using raspberry pi 4b. I use one raspberry pi to ping the router and another pi to collect the CSI. I'm not sure why I don't have any losses.

```
sudo su
makecsiparams -c 36/80 -C 1 -N 1 -m Macaddress
pkill wpa_supplicant
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -vKuABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
ifconfig mon0 up

tcpdump -i wlan0 dst port 5500

```

18563407351, 2020-11-24:
I still have one question. Can the CSI extractor collect CSI of only TCP packets sent from one raspberry pi? Because I want to collect the CSI of packets from one specific host. Is it achievable?

18563407351, 2020-11-25:
@tweigel-dev Are your results similar to the picture I uploaded above? Why only UDP packets captured and no ICMP packet?

zeroby0, 2020-11-25:
> Why only UDP packets captured and no ICMP packet?

Nexmon_CSI collects CSI of all frames, and puts the CSI measurement in a UDP packets, and sends it. You should read the [paper](https://dl.acm.org/doi/pdf/10.1145/3349623.3355477?casa_token=meS1iTZRIkkAAAAA:foLDpCYWtz99stbXfcsAGPgOlRMlZAW3nymx7LZe8rnq1PpYcUX7BfUTSRPhFiEbkf0BGTs6JcJB), it explains the process well.

No matter if the frames are UDP or TCP, the CSI of them is _measured_, and the measurement is put in a new UDP packet in the wifi chip. Those artificially created UDP packets embedding the CSI data is what tcpdump is collecting. The CSI data of ICMP packets should be inside those UDP packets too.

> so how can I know the channel the command Ping is using?

If your wifi router is set to channel 36, then all the WiFi traffic of the router and devices connected to it happen in channel 36. Individual commands like ping cannot choose which channel to send their traffic in.

> Can the CSI extractor collect CSI of only TCP packets sent from one raspberry pi?

You can use the `-m` filter with makecsiparams to capture only packets from that raspberry pi. But they need not be only TCP, any packets originating from that raspberry pi will be collected. More information in the [readme](https://github.com/seemoo-lab/nexmon_csi).

Please go through the readme, the paper I attached, and the [issues](https://github.com/seemoo-lab/nexmon_csi/issues?q=) in this repository; it's the fastest and easiest way to get a good understanding of how it works and of wifi, and linux.

18563407351, 2020-11-25:
> Nexmon_CSI collects CSI of all frames, and puts the CSI measurement in a UDP packets, and sends it. You should read the [...]

I got it! Can't thank you enough!

tweigel-dev, 2020-11-25:
> @tweigel-dev Are your results similar to the picture I uploaded above? Why only UDP packets captured and no ICMP packet?

yes i gather the same output. I think your ICMP question is allready answered :)

18563407351, 2020-11-25:
> > @tweigel-dev Are your results similar to the picture I uploaded above? Why only UDP packets captured and no ICMP packet?
> 
> yes i gather the same output. I think your ICMP question is allready answered :)

I understood. Thanks.